bigtreetech/BIGTREETECH-TouchScreenFirmware #2301: TFT35 E3 V3 - No response at all to any FW?

TheGadge, 2021-12-26:
BTT TFT35 E3 V3
Skr mini E3 v2

- I can't seem to update in any way (or even get a response from screen) that sd or USB has a bin file in.
- (Yet I can see in touch screen mode that U disk is inserted etc)
- I wanted to try the noscreentouch firmware also, but can't even read prev fw releases from BTT.

I've noticed there are bootloader 'fixes' but with zero notes or instructions in a true engineer style lol.

Any ideas, please? 
Many thanks
William

sarvenn, 2021-12-26:
Would you please try the latest marlin bugfix 2.
Yesterday I had a similar issue after uploadinh fw.
Today, I uploaded the latest bugfix and issue is solved.

TheGadge, 2021-12-26:
> Would you please try the latest marlin bugfix 2.
> Yesterday I had a similar issue after uploadinh fw.
> Today, I uploaded the latest bugfix and issue is solved.

Small problem. I’m on klipper. I don’t think it’s that lol. Thanks for trying for me!

radek8, 2021-12-26:
What update do you have a problem with?
SKR or TFT?
The description of your problem is difficult for me to understand.

TheGadge, 2021-12-26:
> What update do you have a problem with? SKR or TFT? The description of your problem is difficult for me to understand.

Oh ok sorry, but the Title is "TFT35 E3 V3 - No response at all to any FW"
No matter what bin file I add to a fat32 sd or USB, the TFT35 gives no response whatsoever, yet detects the card/usb being in.

radek8, 2021-12-26:
So you can't update TFT?
The SD card, not USB, is used for the update.
Cards larger than 32GB may not work well.
Use an 8GB or smaller SD card, formatted to FAT32.
Copy BIGTREE_TFT35_V3.0_E3.27.x.bin, config.ini and folder TFT35 to the card.
After a successful update, the BIN file is renamed to CUR.

TheGadge, 2021-12-26:
> So you can't update TFT? The SD card, not USB, is used for the update. Cards larger than 32GB may not work well. Use an 8GB or smaller SD card, formatted to FAT32. Copy BIGTREE_TFT35_V3.0_E3.27.x.bin, config.ini and folder TFT35 to the card. After a successful update, the BIN file is renamed to CUR.
> 
> Sometimes it helps to reformat the SD Card.

Sandisk extreme 16GB SD card used. 
done all the above. It’s as if the port isn’t even there.

radek8, 2021-12-26:
Format it again.
If that doesn't work, use another card.

TheGadge, 2021-12-26:
> Format it again. If that doesn't work, use another card.

Well, I'm on my 3rd SD card, very strange, but we have action! Many thanks! :)

TheGadge, 2021-12-26:
> Format it again. If that doesn't work, use another card.

Sorted! Many many thanks, I can't believe I've got cards that seem fine in Windows, checked in camera etc, BTT only likes one of them, all between 8-16GB. Oh well!

I've just also update notouch firmware for klipper, and it works a treat after config changes.
I owe you a beer for taking me back to the basics.

radek8, 2021-12-26:
I also have experience with this strange behavior.
I use 2 SD cards, both work and update the FW.
Suddenly, one SD card stops updating the FW.
It can be used to update again in a week. I didn't figure out what it was, but if one doesn't work I'll take the other ...

radek8, 2021-12-26:
Close the ticket if the problem is solved.

oldman4U, 2021-12-29:
Gadge

Please help the community and close the ticket. Thank you

closed 2022-01-05 by bigtreetech