bigtreetech/BIGTREETECH-TouchScreenFirmware #2303: [BUG] Vs Code Complie Error at Master

lnfactionfreddy, 2021-12-27:
Vs Code Complie Error at Master 
### Description

Processing BIGTREE_TFT70_V3_0 (board: STM32F407VG_0x8000; platform: ststm32; framework: cmsis)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Generating firmware...
Done.
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F407VG_0x8000.html
PLATFORM: ST STM32 (15.1.0) > STM32F407VG (192k RAM. 1024k - 32K Flash)
HARDWARE: STM32F407VGT6 168MHz, 192KB RAM, 992KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-cmsis 2.50501.200527 (5.5.1)
 - framework-cmsis-stm32f4 2.6.4
 - tool-ldscripts-ststm32 0.2.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find the default startup file for stm32f407vgt6. Ignore this warning if the startup code is part of your project.
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generate language_xx.ini files:
Processing: language_am.h: Total keywords found:326, File generated:language_am.ini
Processing: language_ca.h: Total keywords found:326, File generated:language_ca.ini
Processing: language_cn.h: Total keywords found:326, File generated:language_cn.ini
Processing: language_cz.h: Total keywords found:326, File generated:language_cz.ini
Processing: language_de.h: Total keywords found:326, File generated:language_de.ini
Processing: language_du.h: Total keywords found:312, File generated:language_du.ini
Processing: language_en.h: Total keywords found:326, File generated:language_en.ini
Processing: language_es.h: Total keywords found:326, File generated:language_es.ini
Processing: language_fr.h: Total keywords found:326, File generated:language_fr.ini
Processing: language_gr.h: Total keywords found:326, File generated:language_gr.ini
Processing: language_hu.h: Total keywords found:326, File generated:language_hu.ini
Processing: language_it.h: Total keywords found:326, File generated:language_it.ini
Processing: language_jp.h: Total keywords found:326, File generated:language_jp.ini
Processing: language_nl.h: Total keywords found:326, File generated:language_nl.ini
Processing: language_pl.h: Total keywords found:326, File generated:language_pl.ini
Processing: language_pt.h: Total keywords found:326, File generated:language_pt.ini
Processing: language_ru.h: Total keywords found:326, File generated:language_ru.ini
Processing: language_sk.h: Total keywords found:326, File generated:language_sk.ini
Processing: language_sl.h: Total keywords found:326, File generated:language_sl.ini
Processing: language_tc.h: Total keywords found:326, File generated:language_tc.ini
Processing: language_tr.h: Total keywords found:326, File generated:language_tr.ini
Processing: language_uk.h: Total keywords found:326, File generated:language_uk.ini
Total language files processed: 22
Compiling .pio\build\BIGTREE_TFT70_V3_0\FrameworkCMSIS\system_stm32f4xx.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\Startup\stm32f40_41x\startup_stm32f407xx.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\base64\base64.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\cmsis\stm32f4xx\system_stm32f4xx.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\misc.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_adc.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_can.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_cec.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_crc.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_cryp.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_cryp_aes.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_cryp_des.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_cryp_tdes.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dac.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dbgmcu.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dcmi.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dfsdm.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dma.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dma2d.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_dsi.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_exti.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_flash.o
Compiling .pio\build\BIGTREE_TFT70_V3_0\src\src\Libraries\fwlib\stm32f4xx\src\stm32f4xx_flash_ramfunc.o
[... 82 lines cut ...]
                 from TFT\src\User\API\Colors.c:1:
TFT\src\User\API/LCD_Colors.h:53:3: note: previous definition of 'LCD_GREEN' was here
   LCD_GREEN,
   ^~~~~~~~~
In file included from TFT\src\User\API\Colors.c:1:0:
TFT\src\User\API\Colors.h:51:3: error: redeclaration of enumerator 'LCD_BLUE'
   LCD_BLUE,
   ^~~~~~~~
In file included from TFT\src\User/includes.h:83:0,
                 from TFT\src\User\API\Colors.h:9,
                 from TFT\src\User\API\Colors.c:1:
TFT\src\User\API/LCD_Colors.h:54:3: note: previous definition of 'LCD_BLUE' was here
   LCD_BLUE,
   ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_TFT70_V3_0\src\src\User\API\Colors.o] Error 1
====================================================================================== [FAILED] Took 9.51 seconds ======================================================================================
Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT70_V3_0  FAILED    00:00:09.512
================================================================================ 1 failed, 0 succeeded in 00:00:09.512 ================================================================================

### Hardware Variant

BTT TFT 70

Guilouz, 2021-12-28:
No issue for me to compile it for TFT70

radek8, 2021-12-29:
For me, the compilation was also successful.
The error will be in your edits or you have combined files from different versions.

closed 2022-01-05 by bigtreetech