bigtreetech/BIGTREETECH-TouchScreenFirmware #2304: [BUG] Buggy console, infinite loop after G29

rondlh, 2021-12-28:
### Description

Using the console running a G29 causes an infinite loop

### Steps to reproduce

1. Start fresh
2. In the console first run G28 (works fine) then run G29 which causes an infinite loop

**Expected behavior**
I expected to see the G29 output of a 5x5 bed probing.

**Actual behavior**
Infinite loop, I see the page counter going crazy, it seem to be spinning through the pages continuously. The touchscreen does not respond to commands anymore.

### Hardware Variant
BTT TFT35 V3.0 with MKS SGEN_L 2.0 board with Marlin 2.0.9.3, Serial communication at 1M baud.

### TFT Firmware Version & Main Board Firmware details
Latest firmware (Dec 26th 2021)

### Additional Information
If I run a G29 via the serial port of the motherboard then I can see the output generated as follows (slight customized Marlin output)

G29
>>> PROBING BED...
Bilinear Leveling Grid:
      0      1      2      3      4
(TOP-LEFT)
 4  -0.122   -0.143  (-0.145)  -0.124   -0.131 
 3  -0.052   -0.077   -0.068   -0.063   -0.059 
 2  -0.004   -0.023   -0.014   -0.026   -0.039 
 1  -0.007   -0.024   -0.013   -0.020  [-0.001]
 0  -0.082   -0.083   -0.066   -0.049   -0.030 
Max: [-0.001mm]   Min: (-0.145mm)
Difference:0.144mm   Average:-0.0586
Z-offset: -2.2500
X:190.5000 Y:148.5000 Z:7.2653 E:0.0000 Count X:15240 Y:11880 Z:23200
//action:prompt_end
//action:notification Bed leveling done.

digant73, 2021-12-29:
are you sure you are on the latest TFT fw? that kind of bug was fixed some months ago and I don't have this issue anymore

rondlh, 2021-12-29:
> are you sure you are on the latest TFT fw? that kind of bug was fixed some months ago and I don't have this issue anymore

Yes, I'm sure, I use the latest version released on December 24th 2021, with support for vertical displays. If I check the version it only says TFT35_V3.0.25.x Dec 25 2021. I found more erratic behavior of the console, but this G29 bug is repeatable. I will try to find more examples.

oldman4U, 2021-12-30:
Has to be v27.x and not 25.

What is the benefit using such high baud rates?

rondlh, 2021-12-30:
> Has to be v27.x and not 25.
You are right, it is v27.x
> What is the benefit using such high baud rates?
1Mbit helps to keep the motion queue full at all times during printing, it's probably overkill, but it works fine and is supported by the TFT and Marlin (SW+HW).

I just tested G29 on 115200 and 250000 baud, this does not solve the issue.

rondlh, 2021-12-30:
I did some more tests at lower baud rates and other commands. Lower baud rates doesn't solve the problem.
G29 (bed leveling) --> Infinite loop, page counter goes crazy, screen does not respond anymore to touches.
M43 (show pins) --> Only 2 pages are shown, all text is turned orange, touch screen works.
M503 (report settings) --> No problem
M122 (TMC status) --> No problem
M119 (endstop status) --> No problem

digant73, 2022-01-01:
@rondlh try to reduce baudarate with mainboard. 1M? try 250K, 125K

rondlh, 2022-01-02:
> @rondlh try to reduce baudarate with mainboard. 1M? try 250K, 125K

@digant73 I tried, both 250K and 115200, but this did not solve the problem unfortunately.

digant73, 2022-01-04:
@rondlh can you provide your Configuration.h and config.ini files?

rondlh, 2022-01-09:
I put the Configuration.h and config.ini files in the attached Zip file
[CONFIG.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7834552/CONFIG.zip)

digant73, 2022-01-09:
@rondlh I applied your setting but no loop in my case, although I have UBL in Marlin. I also see that you enabled MBL in Configuration.h while it seems you have BBL in your Marlin. If you have BBL you should set BED_LEVELING_TYPE to 3 in Configuration.h (or 1 for auto-detect). Also you have two ports enabled. You could try test G29 with only the first port enabled

rondlh, 2022-01-10:
@digant73 Thanks a lot for your efforts, I will do a few more tests. You are right about the TFT bed leveling setting, it should be BBL. I have an ESP3D on the second port which I use now to give the G29 command, disabling it might give more insight into this issue. I will report back... thank!

digant73, 2022-01-11:
I did also those tests with no loop. With M43 I also obtain a partial output (the last part of the output). But this could be related because the terminal menu is open after a while (and the initial part of the M43 output was not catched by the terminal). Same result with Wifi port enabled. In both cases, only the command M43 is in orange while the output is white as expected

rondlh, 2022-01-11:
@digant73 Thanks again for your feedback! The console result for G29 didn't change after changing to BBL or disabling the Wifi port. I only saw a difference in the response of M43 when disabling Wifi. 
Currently this topic is not so hot for me as I can use Wifi to give commands and catch the console output without problems. 
I will do some more test to figure out what is going on and report back.

rondlh, 2022-01-19:
I did some tests, I increased TERMINAL_MAX_CHAR to 6800 (terminal.c), then the problem disappears and 18 pages are available and the coloring of the text is as expected (only M43 in orange). In my case M43 results in a response of 6004 bytes , my TFT (BTT TFT35V1.0) has 96kB of ram, which results in 96 * 45 = 4320 bytes of terminal space.
This also solves the infinite loop problem I found with G29.

digant73, 2022-01-20:
yes, the issue seems caused when the buffer terminalBuf (having size TERMINAL_MAX_CHAR) is reached. When I have some time I will have a look to Terminal

digant73, 2022-01-23:
@rondlh please, if you have some time test PR #2345. It should be solve at least the infinite loop etc... As you know, the partial list for M43 is due to limited buffer size

rondlh, 2022-01-24:
@digant73 Tested, it solves all my issues, great job, thanks!
BTW: In my view the page up/down can loop around, it's more flexible that way (for example: page 10 press up --> page 1, page 1 press down --> page 10)

digant73, 2022-01-24:
Good it solved all your issues. The handling of the pages is more complicated than we can imagine.
Your bug report will be automatically closed once the PR is merged into the master branch  (just to keep track of the thread). So you don't need to close it

rondlh, 2022-01-24:
I played around a bit with the page counter loop around, and came up with this implementation which seems to work fine (terminal.c) 2 else cases added.

```
      case TERM_PAGE_UP:  // page up
        if (terminalData->pageIndex < terminalData->pageCount)
          terminalData->pageIndex++;
	else
	  terminalData->pageIndex = 0; // IRON, added, loop around
        break;

      case TERM_PAGE_DOWN:  // page down
        if (terminalData->pageIndex > 0)
          terminalData->pageIndex--;
	else
	  terminalData->pageIndex = terminalData->pageCount; // IRON, added, loop around
        break;
```

PR #2345 (2022-01-25): Bugfixes in Terminal + cleanup
**BUGFIXES:**
* Fixed some bugs in Terminal: Terminal had some general bugs such as wrong rolling pages, wrong page count, wrong colors (e.g. orange for ACK messages), the prev button increased the page (instead of decreasing it), next button decreased the page (instead of increasing it), when the buffer was rolled the page count was reset to 1, TFT freeze (infinite loop) if character '\0' was present in the buffer etc. It fixes also #2304.
* Minor fix on LEDColor menu: LED color was not changed when increasing/decreasing an RGB component not having the focus. Al [... cut]
closed 2022-01-25 by bigtreetech