bigtreetech/BIGTREETECH-TouchScreenFirmware #2305: [Q] BTT-TouchScreenFirmware on the MKS TFT35, anybody interested?

rondlh, 2021-12-29:
With the of Darkspr1te (https://github.com/darkspr1te/BIGTREETECH-TouchScreenFirmware) I managed to get the BTT TouchScreenFirmware to work well on the MKS TFT35. I managed to solve the unreliable touchscreen issues he found by slowing down the SPI bus during touch sampling to speeds within the range of the touch screen processor to solve this issue. The required changes/additions are not very big, about 10 #ifdef statements left and right. Is this of interest here?

digant73, 2022-01-01:
provide here the new/changed files (e.g. platformio, pin_MKS_TFT35xxx.h etc...)

rondlh, 2022-01-02:
I attached a zip file with all the modified and additional files. Please note that MKS TFT uses encryption of the firmware. A encrypted and non-encrypted firmware file are created when building, the encrypted fw (default) has the correct name for direct upload to the TFT. I personally use a bootloader without encryption, so I use the unencrypted fw file.
Also note that the touchscreen calibration data and user settings are stored in flash, which is cleared when flashing a new firmware, so you lose your settings and will be prompted to calibrate the screen again after firmware update.
Please let me know if you have any questions.

[BTT TouchscreenFirmware Dec 30 2021 - MKS TFT35 V1.0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7798677/BTT.TouchscreenFirmware.Dec.30.2021.-.MKS.TFT35.V1.0.zip)

digant73, 2022-01-02:
@rondlh ok, understood. I will integrate it in a further PR

digant73, 2022-01-14:
@rondlh Is MKSTFT35_V1_0.27.x.bin (copied to folder "Copy to SD Card root directory to update") the encrypted version and MKSTFT35.BIN (in .pio folder) the not encrypted version? That should be the correct way

Sanerok, 2022-02-10:
@rondlh , As mentioned here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2320#issuecomment-1035204589
I tried to flash my tft with your files, seems that it does not work. 
I also tried to flash encrypted bin from your [BTT TouchscreenFirmware Dec 30 2021 - MKS TFT35 V1.0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7798677/BTT.TouchscreenFirmware.Dec.30.2021.-.MKS.TFT35.V1.0.zip) and ended with the same result.
If you are using a bootloader without encryption, so maybe the problem is in the encrypted file? Could you help me find a bootloader without an encryption, so I can check the firmware without encryption? Also, if it is necessary to flash the bootloader using stlink, could you tell which address should I flash it to ? 

Thanks in advance.

PR #2397 (2022-02-16): Added MKSTFT35 support + cleanup
**IMPROVEMENTS:**
* Added MKSTFT35 support
* Some code reduction: Added functions getFoldername and getFilename in vfs.h / .c in order to be reused in Printing.c, Print.c and PrintingMenu.c
* Reviewed README.md: Added more details and sections, applied proper paragraph styles etc.
* Minor cleanup

resolves  #2305

**PR STATE:** Ready for merge
closed 2022-02-16 by bigtreetech