bigtreetech/BIGTREETECH-TouchScreenFirmware #2306: [Q] (Your Question?)

AcidSubway, 2021-12-30:
HI all,

first of all, thanks for the awesome firmware.
It runs very well so far :D
I have an Artillery SW1 and have flashed me the software 1.27P7. To the display unit you can solder a MKS WIFI module, this I have also done. However, the WebServer gets no connection, I have activated Wifi in the settings and tried any connection speed, but without success.
I have flashed this WebServer on my module:
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/wiki/MKS_Robin_WIFI#update-the-esp3d-firmware-to-mks-robin-wifi
What am I doing wrong?

Best g.

oldman4U, 2021-12-30:
The title of the ticket is wrong for sure. What kind of firmware related issue do you expect causes the issue? Because if you believe that this is a user group where you can post a problem- I am sorry - you are also wrong. Because this is the BTT TFT firmware repository and not a user group. 

Regards

rondlh, 2021-12-31:
The TFT will work fine with any ESP module running the ESP3D firmware that is correctly wired up, make sure to enable the correct serial port, and set the serial communication on both the ESP and TFT to the same speed. Motherboard and ESP3D serial communication speed can be configured independently in the latest BTT firmware. An ESP with the MKS Wifi firmware will _not_ work with the TFT, but you can flash the ESP3D firmware to an MKS Wifi module.

AcidSubway, 2022-01-01:
thanks for the info, that sounds good.
Do you have a link for me where the ESP3D is already packed in a .bin?

AcidSubway, 2022-01-01:
if I do everything as in the video, is it right? Then I get my module to run?

https://www.youtube.com/watch?v=NGgzw-XayEo

AcidSubway, 2022-01-02:
i have now installed esp3D but can't get any proper communication. If I adjust the ports in the tft for WIFI then the temperatures come, but everything else does not work. I can not move a printer :/ can someone help me here?

rondlh, 2022-01-03:
@AcidSubway If you see the temperatures then you are on the right way, it at least means you are receiving serial data. Send "M119" via the ESP3D command console and see if you get a response from the motherboard about the endstop status. M20 is used to list storage, which is not always supported, depending on your configuration. I guess this should work because you got an error message generated by the motherboard, so if you still have issues then this is just the motherboard not responding to your commands as expected.

MacarioPatrick, 2022-01-03:
Maybe I should solder a Wi-Fi module onto the TFT as well.

rondlh, 2022-01-04:
> Maybe I should solder a Wi-Fi module onto the TFT as well.

Plug it in, don't solder it.

AcidSubway, 2022-01-05:
Hello, yes that is a good idea.
in the meantime I have installed ESP3D V3.0 on the chip. now it works much better, without problems you can see everything in the WebServer. 
[image]
[image]

bigtreetech, 2022-01-05:
Has your issue been solved now? I will close this issue. You can reopen this issue at any time if you have any unresolved issues.

closed 2022-01-05 by bigtreetech

AcidSubway, 2022-01-05:
Thank you, it can be closed. everything works.