bigtreetech/BIGTREETECH-TouchScreenFirmware #2307: [BUG] Incorrect sort order of files on USB/SD

rondlh, 2022-01-02:
I noticed that the file order displayed is not according to my settings: files_sort_by:0
#   Options: [Date Newest First: 0, Date Oldest First: 1, Name Ascending: 2, Name Descending: 3]

### Description
The order of files shown on the display does not match the "Newest First" file order.

### Steps to reproduce
1. Copy some files to USB or SDCARD with older dates
2. Make sure the configuration is set to Newst First or Oldest First
3. Check if the displayed order on the display is correct.

**Expected behavior**
Get the correct file order, newest first.

**Actual behavior**
The files are in the wrong order. It's not Newest first or oldest first, or even Name Ascending/Descending.
This both happens on the SDCard and USB drive.

### Hardware Variant
BTT TFT35 V1.0

### TFT Firmware Version & Main Board Firmware details
Latest firmware (Dec 30 2021)

### Additional Information
I formatted the USB drive and SDCard and copied the same files again, the problem persisted.
I attached a picture showing the Windows Newest first explorer view, which shows the correct order, and the file order shown on the display.
[image]
[image]

sarvenn, 2022-01-02:
Could you add "Date Created" to your windows explorer page? I'm not a developer but I know that date modified and date created are two different things. That feature might be looking for the date created values, I don't know but it worths to try.

rondlh, 2022-01-02:
> Could you add "Date Created" to your windows explorer page? I'm not a developer but I know that date modified and date created are two different things. That feature might be looking for the date created values, I don't know but it worths to try.

Good point! Like stated, I formatted the USB drive and SDCARD and copied the gcode files again, all creation dates/times are the same: 2/1/2022 6:00 PM. If I click on the column header for "Date created" to sort the files, then the order of the files does not change compared to sorting on "Date modified". An updated explorer view is attached, also a small archive of the files are attached. Perhaps someone can copy the files to their card and check the file order on the display.
[image]

[GCODE_FILES.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7799059/GCODE_FILES.zip)
.

rondlh, 2022-01-02:
I think I found and solved the problem, it's an integer rollover issue.
The file/folder date and time index are stored together in one 32 bit (signed) integer in myfatfs.c:
```
folderDate[infoFile.folderCount] = (finfo.fdate * 100000) + finfo.ftime;
fileDate[infoFile.fileCount] = (finfo.fdate * 100000) + finfo.ftime;
```
The used day multiplier (100000) is too big because the time counter only goes to 49021 (fat fs last write date), so no need to use 100000, you could use 49022 or an easy to calculate multiplier like 1<<16 (=65536).

The fileData array is a 32bit Integer, an unsigned integer would be better.
I updated the following in myfatfs.c to solve the issue:

```
bool scanPrintFilesFatFs(void)
{
  FILINFO finfo;
  uint16_t len = 0;
  DIR dir;
  uint32_t folderDate[FILE_NUM]; // Changed to unsigned
  uint32_t fileDate[FILE_NUM];   // Changed to unsigned

This data is forwarded to the "compareFile" function, which also must be updated:
```

`bool compareFile(char * name1, uint32_t date1, char * name2, uint32_t date2)`

The next rollover will be in about 120 years. I attached the updated myfatfs.c
[myfatfs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7799258/myfatfs.zip)

digant73, 2022-01-02:
good work. I have an open PR, so I can add also this fix unless you are already opening a PR

PR #2308 (2022-01-04): Fixed #2300
**BUGFIXES:**
* Fixed a bug reported in #2300: Missing thumbnail icon's when going back/forth pages.
* Fixed #2307

fixes #2307 (bug found and solved by @rondlh)

**PR STATE:** Ready to merge
closed 2022-01-04 by bigtreetech