bigtreetech/BIGTREETECH-TouchScreenFirmware #2309: [Q] Bin file for Biqu B1?

ramosglauco, 2022-01-04:
Hi. Sorry if my question seems silly. I have a Biqu B1 with Marlin 2.0.7.2 installed. I have a question about which bin file I need to upload exactly:

BIGTREE_TFT35_V3.0.27.x.bin
BIGTREE_TFT35_V3.0_B1.27.x.bin
BIGTREE_TFT35_V3.0_E3.27.x.bin

I'm attaching a TFT photo. Please, can you help me?

[image]

rondlh, 2022-01-04:
BIGTREE_TFT35_V3.0_B1.27.x.bin should be the correct firmware.
The TFT will only flash if it finds the correct firmware filename, so don't rename the file!
Either way, you will not damage or brick the TFT if you flash the wrong firmware...

oldman4U, 2022-01-04:
ReadMe has all the information.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2022-01-05 by bigtreetech