bigtreetech/BIGTREETECH-TouchScreenFirmware #2310: second nozzle does not heat past 200 on touch screen

manganD, 2022-01-04:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

manganD, 2022-01-05:
no its at 275 , should i have ?

manganD, 2022-01-05:
i am so sorry i found the solution . i went back and looked at my config.ini file  to double check the temp setting i saw that while i changed the number of extruders i missed the hotend line . i changed the hotend to 2 and now everything works as it should . i do apologize

closed 2022-01-05 by manganD

gordo3di, 2022-01-05:
> i am so sorry i found the solution . i went back and looked at my config.ini file to double check the temp setting i saw that while i changed the number of extruders i missed the hotend line . i changed the hotend to 2 and now everything works as it should . i do apologize

Been there

manganD, 2022-01-05:
No I didn’t change the max temp its at 275. Should I have ?i checked when I saw you comment and  I did however see that I missed the hotend line  and didn’t change that to 2 . I changed the extruders to 2 but had missed the hotend line.  Ive since changed it and not the temps are no longer limited to 200 and I can select each hotend on the touchscreen just like I could with the marlin screen. So everything is working as it should  Sent from Mail for Windows From: gordo3diSent: Wednesday, January 5, 2022 9:25 AMTo: bigtreetech/BIGTREETECH-TouchScreenFirmwareCc: manganD; AuthorSubject: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] second nozzle does not heat past 200 on touch screen (Issue #2310) Did you change the max temp in config.ini as well?