bigtreetech/BIGTREETECH-TouchScreenFirmware #2311: Does command M605 S3 (mirror mode) fail?

angeldrug, 2022-01-06:
Hello. Can you tell me why the M605 S3 (mirror mode) command is not executed in TFT mode?
Printer with two extruders. Created four commands in the custom menu:
M605 S0 (Full control mode)
M605 S1 (AUTO_PARK mode)
M605 S2 (DUPLICATION mode)
M605 S3 (MIRRORED mode)

The commands M605 S0, M605 S1, M605 S2-work and switch modes.
Command M605 S3 - does not work and mirror mode does not turn on.
If you send the M605 S3 command through the terminal, it also does not work and the mirror mode does not turn on.
In marlin mode, through the IDEX menu, all commands are executed and work

radek8, 2022-01-07:
I don't see the S3 switch for Marlin documented anywhere for the M605

https://marlinfw.org/docs/gcode/M605.html

https://reprap.org/wiki/G-code#M605:_Set_dual_x-carriage_movement_mode

angeldrug, 2022-01-07:
radek8 Marlin \ src \ gcode \ control \ M605.cpp - you can read here

rondlh, 2022-01-07:
The TFT does not modify the M605 command in any way, it just forwards the command to the motherboard. This does not seem to be an TFT issue.

angeldrug, 2022-01-07:
Yes, I think so too. I sent a command from Repetier-Host and it doesn’t work either. I’ll ask someone in the marlin section. Thank you.

radek8, 2022-01-07:
File: Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/control/M605.cpp
Incidentally, line break is missing on line 103; ?

[image]

angeldrug, 2022-01-07:
I am not a programmer either. That is why I am asking here. The printer had a board from MKS and I replaced it with SKR PRO. And now I cannot understand which command the mirror mode is turned on.

radek8, 2022-01-07:
I sent a question to Marlin

oldman4U, 2022-01-07:
@rondlh Please check #2304

closed 2022-01-08 by angeldrug