bigtreetech/BIGTREETECH-TouchScreenFirmware #2313: [BUG] Error in PIDTUNE? TFT50v3 / Wrong PIN for HOTBED is used

SpRiNgToIfEl, 2022-01-08:
It looks like the current firmware (TFT) has a problem with the PIDTUNE.
I manually stored a different PIN (PB0) for the HOTBED in Marlin FW 2.0.9.3 (and before).

If I now start PIDTUNING via the TFT50v3, the HOTEND (PB1) will be tested correctly, but when PIDTUNING the HOTBED, the apparently wrong PIN (PD12) is activated (to be seen on the LED on the SKR PRO 1.2).

If I enter the command "M303 E-1 C3 S70" in PRONTAFACE 2.0.0rc8 or Octoprint, the correct PIN is addressed / used, i.e. the HOTBED (PB0).

Is there a way to control the PINS of the TFT-FW?
Which PINS are controlled by the TFT50v3?
Which file of the TFT-FW has to be edited?

I have always run the PIDTUNE via the terminal, now I wanted to run it for the first time via the TFT, unfortunately the wrong PIN is addressed / tested on the HOTBED.

### Steps to reproduce

On the TFT50v3 enter the PIDTUNE an Test the HOTBED >>> Wrong PIN is tested.
Enter the PIDTUNE (M303 E-1...) via terminal, the correct PIN is used.

**Expected behavior**

**Actual behavior**

### Hardware Variant
BTT TFT 50v3 with FW Vx.x.27 (10.04.2021)
BTT SKR PRO 1.2 with FW Marlin 2..0.9.3 (and before)

rondlh, 2022-01-08:
1. I recommend to upgrade to the latest version first and try again.
2. The TFT is not aware of pins or hardware details, it only talks to the motherboard in Gcode. So if Marlin is configured correctly then the TFT will send some Gcode to the motherboard and then it's up to Marlin.
3. For the Bed autotune the TFT will send the following to the motherboard: M303 U1 C8 E-1 S<set bed temp>. Also the part cooling fan will be set to the maximum speed (M106 S255), this might switch on some LEDs on your motherboard...
4. The TFT has a console where you can type in you own command, you could try your command there (M303 E-1 C3 S70)

SpRiNgToIfEl, 2022-01-08:
Thanks for the message.

Today I downloaded Marlin 2.9.0.1, configured it and did a PIDTUNE using the TFT50v3.
The TFT works as it should with version 2.9.0.1, but not with version 2.9.0.3.
I assume that there is a bug in the current marlinFW (the nightly version of 2.9.0.3 is also affected).

rondlh, 2022-01-08:
I think you got the Marlin version numbers a bit wrong, we are currently at Marlin 2.0.9.3.
Looking forward to your console test results...

Wombat37, 2022-01-09:
This looks very similar to an issue I reported last year - #1856.

Take a look at that and see if what you're seeing is the same.

Unfortunately, despite my optimism and efforts to prevent it, the issue was abandoned for the second time yesterday (same day you raised your issue) because of a lack of attention. Looks like we're stuck with this problem.

I did do a lot of investigation for #1856 and you'll find work-arounds in that posting.

Good luck

rondlh, 2022-01-10:
@Wombat37 I'm sure that if you provide correct feedback your problem will be resolved within a few days. SpRiNgToIfEl problem is based on a misunderstanding of the Touchscreen firmware. Either way, I'm waiting for the test results...

SpRiNgToIfEl, 2022-01-10:
short feedback:

- printer started on next day (marlin 2.9.0.3 (also nightly))
- TFT50v3> menu> settings> machine> tuning> pid> hotbed> wrong PIN (PD14) is tested!

### printer reboot via reset button ###

- TFT50v3> menu> Settings> terminal> m303 e-1 c3 s80> the correct pin (PB0) for pidtuning is addressed
- reboot via reset button
- TFT50v3> menu> settings> machine> tuning> pid> hotbed> the correct PIN is tested!

Strange!

SpRiNgToIfEl, 2022-01-10:
> I think you got the Marlin version numbers a bit wrong, we are currently at Marlin 2.0.9.3. Looking forward to your console test results...

I have tested the following versions:
(from new to old to find the mistake)
2.9.0.1
2.9.0.2
2.9.0.3

nightly version as well ...

All versions have the same problem.

digant73, 2022-01-11:
@SpRiNgToIfEl did you reset EEPROM from Machine->EEPROM menu after you install Marlin fw?

oldman4U, 2022-01-14:
What is the status of the reported issue?

SpRiNgToIfEl, 2022-01-15:
oh sorry, i'm on vacation right now, will give feedback in the coming week.

oldman4U, 2022-02-02:
What is the status of the reported issue?

Please help the community and close the ticket once you do not need it anymore. Thank you

SpRiNgToIfEl, 2022-02-03:
> What is the status of the reported issue?
> 
> Please help the community and close the ticket once you do not need it anymore. Thank you

I have now reloaded, configured and flashed the latest Marlin 2.0.9.3 as well as the latest Nightly. Result: Hotbed pid tuning via the TFT ends in thermal runaway.

I have tested both thermistors 1 and 30 (KIS3D) in configuration.h, no differences. There is an isolation under the hotbed.

SpRiNgToIfEl, 2022-02-03:
> @SpRiNgToIfEl did you reset EEPROM from Machine->EEPROM menu after you install Marlin fw? Also, from the console try M303 U1 C8 E-1 S80. This is the command sent by the PID menu (see the additional U1 parameter and C8 instead of C3)

i always use M502 + M500 after flashing

radek8, 2022-02-03:
Be careful when transferring configurations between Marlin versions
If you are testing multiple versions of Marlin, you must use the configuration files for that version, which you will modify according to your configuration.
You must never copy an entire configuration file from one version to another.

SpRiNgToIfEl, 2022-02-03:
Thanks for the tip, but each version of marlin has his own edited config-file. I dont copy it from one version to the other.

digant73, 2022-02-04:
@SpRiNgToIfEl if with gcode M303 U1 C8 E-1 S80 sent from terminal or pronterface you have issues, you should open a ticket on Marlin repo.

digant73, 2022-02-16:
@SpRiNgToIfEl, @bigtreetech I think this issue could be closed. All support from TFT fw side was given. The issue is clearly on mainboard

oldman4U, 2022-02-16:
@Msq001

closed 2022-02-18 by bigtreetech