bigtreetech/BIGTREETECH-TouchScreenFirmware #2316: BTT TFT_E3_V3 and Octopus V1_1 Firmeware need (to much for me)

bmw628de, 2022-01-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

oldman4U, 2022-01-10:
Lieber Jan. Bitte mach ein ordentliches Ticket welches mit der BTT TFT FIRMWARE zu tun hat, oder schließe es wieder.

Danke

bmw628de, 2022-01-10:
Firmware is the Problem
How to compile the Octopus V1.1 Firmware, with stepper TMC2209
The TFT E3 V3 Firmware set the Display mirrored from left to right.
Display may not read the board or the Firmware is wrong.

have any Body a or two Firmwar Files for this stuff. Or with data i need

3 Days i kompile to nothing. I look many Videos, and i hope i get here some help, please help.

rondlh, 2022-01-11:
> Hope its better
Perhaps better, but still not good enough, better close this ticket, this is not the right place for this.

oldman4U, 2022-01-11:
This is the wrong place for all kinds of setup and configuration questions.

What do you mean with mirror? Attach a picture.

The TFT works as soon as power is connected, independent of a connected mainboard.

bmw628de, 2022-01-11:
[image]
  See this

oldman4U, 2022-01-11:
Follow the Firmware Update and after that the troubleshooting section of the ReadMe which you can find at the front page of this repository.

rondlh, 2022-01-11:
@bmw628de It seems you used the wrong firmware, that would explain the mirrored output... flash again with the correct firmware.

radek8, 2022-01-11:
Use a file reset.txt

bmw628de, 2022-01-11:
thanks at all!
nothing helped yesterday! Now i used the reset.txt a second time & its work.
For German People a german Readme.de.txt this is the easy way.
You Helped me i help you

Firmware Update Display für Reset

Beispiel:(TFT35E3V3.0)
1. leere SD Karte Format FAT32
2. Dateien vorher anpassen nach bedarf zum Beispiel
   config.ini 
   baudrate 
   anzahl der Ports 
   Sprache
3. Dateien auf SD Karte Kopieren
   Aus copy to SD Ordner
   aus Language Pack z.b. language_de.ini
   aus Theme Ordner Passendes Display z.b. TFT35
   Bigtree_TFT35_V3.0_E3.27.x.bin

4. Display abschalten 
   SD Karte einfügen

5. Abwarten fertig

End

oldman4U, 2022-01-11:
Perfect. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2022-01-11 by bmw628de