bigtreetech/BIGTREETECH-TouchScreenFirmware #2319: [BUG] Strange behavior when setting the fan.

radek8, 2022-01-13:
I am using fan controllers (M710)
The fan is set to 66%.
If I open the fan menu, the fan (CtA) displays the speed 66/00. It does not respond when you click to lower the temperature. By clicking on the temperature increase, the values are set to 1/1.

If I edit the value (F0) before editing the value of the controller fan (CtA), then I go to the value (CtA) it still behaves badly.
if I leave the menu and return again, Option (CtA) 66/66 is displayed and editing works normally.

The fan setting (F0) also does not work properly.
If I set the fan speed via Gkod (M106 S280)
The value (F0) 50/0 is displayed in the fan menu.
The value cannot be decreased and when increased, it is calculated from the beginning of 1/1.

For clarity, I enclose a link to the video.

https://youtu.be/qP6O3oVJ3eM

rondlh, 2022-01-14:
It will be very difficult to help you if you do not provide basic required information. Please have a look at the bug report template and add the requested information.

radek8, 2022-01-14:
I don't think there's a problem with my configuration. The problem will be in the fan menu logic. The problem is easily reproducible.
For completeness, I will add the information you require in the evening when I am at the printer.

kisslorand, 2022-01-14:
@radek8 
I am on it, will have a fix most probably today.

kisslorand, 2022-01-14:
Check https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2291/commits/a011780e6d6741989fa8916297a77e0beed194e6

radek8, 2022-01-14:
@kisslorand 
Your patch works great.
You're a great programmer.
Thank you.

radek8, 2022-01-14:
@kisslorand 
If you write in the description of your PR:
resolves #2319
This ticket will be tied to your PR. The ticket will be marked with an icon that a fix exists. Your PR will have an icon that solves the problem. After merging, the ticket with the problem will be closed automatically.

kisslorand, 2022-01-25:
There's a new PR (#2349) that fixes this bug.

PR #2349 (2022-01-25): Fix Fan menu odd behaviour
### Requirements

BTT or MKS TFT

### Description

This PR fixes the odd behaviour of the Fan menu.

### Benefits

### Related Issues

Fixes #2319
closed 2022-01-25 by bigtreetech