bigtreetech/BIGTREETECH-TouchScreenFirmware #2321: [BUG] Load and Unload buttons G-Code Error

mpheyse, 2022-01-14:
### Description

After updating  my TFT35 V3.0 E3 with current Firmware ( as of 1/13/2022) from this repository
The now re-labeled "(un)Load" menu's buttons do not work. The both give a G-code error.

### Steps to reproduce

1. upgrade to current firmware
2. bring the Hot-end to temp
3. enter the newly renamed "(un)Load" menu
4. Press the unload or load buttons

**Expected behavior**
Extruder Motor moves 

**Actual behavior**
Screen Pops a G-Code error. Extruder Motor does not run

### Hardware Variant
TFT35 V3.0 E3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x.bin
  Marlin firmware (current as of Oct 21')

### Additional Information
Run-out Menu Buttons do work.  If I drop the Screen into Marlin mode, I can load filament still.  If I then trigger a Filament run-out during a print, it does stop and park Properly.  The Load and unLoad filament buttons on that screen do work, just like the main one had prior to this new firmware. The Problem is I must be in a Print, and trigger a filament fault to get this menu.

kisslorand, 2022-01-14:
Have you read the Readme and enabled FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)?

rondlh, 2022-01-15:
What Gcode error message do you get? Can you be more specific?

radek8, 2022-01-15:
It has the latest FW TFT versions.
I tested the unLoad menu.
Works without problems when printing and out of print.

oldman4U, 2022-01-17:
What is the status of the ticket?

oldman4U, 2022-01-23:
@Msq001

Please check and close. 

Thank you

closed 2022-01-24 by bigtreetech