bigtreetech/BIGTREETECH-TouchScreenFirmware #2328: [BUG] Selecting White LED turns the screen black

seth995, 2022-01-18:
### Description

Any time the LED turns white (either via the LED Color selection menu or via the Event LED) the TFT screen will turn black.  Touch events still work.

### Steps to reproduce

1. Navigate to Menu>Settings>Machine>LED Color
2. Select WHITE

**Expected behavior**
White NEO_Pixel lights illuminate
Screen stays on and acts with normal operation

**Actual behavior**
White NEO_Pixel lights illuminate
Screen Turns black

### Hardware Variant
TFT35 e3 V3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x
Marlin 2.0.9.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[ConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7889803/ConfigFiles.zip)

digant73, 2022-01-19:
LEDColor menu sends the following gcode for white color:
M150 R255 U255 B255 W0 P255 I255.

Try to send it from Pronterface and see what happens

Guilouz, 2022-01-19:
No issue for me, manually or with event LED.

seth995, 2022-01-20:
Thank you for investigating.  Sending via pronterface has the same problem, but modifying the brightness P255 to P200 leaves the screen on, but non functional.  Changing to P127, everything works fine.  This feels like a power issue and driving too many LEDS off of the board as white draws more power then other colors.  
Thank you again, Closing the bug.

closed 2022-01-20 by seth995