bigtreetech/BIGTREETECH-TouchScreenFirmware #2329: [BUG] During Mesh Validation printer crashes and restarts

seth995, 2022-01-18:
### Description

Using the Mesh Validation feature, during print the printer reboots

### Steps to reproduce

1. Navigate to Mesh validation Menu > Movement> Bed Level
2. Press Mesh Valid.
3. select PLA to set heater (this is the only one I have tested with)
4. Allow printer to heat and start printing

**Expected behavior**
Expect mesh validation print to complete with out error

**Actual behavior**
on 3rd box/circle on the right side of the bed, the printer reboots

### Hardware Variant
TFT35 E3 V3.0, SKR Mini E3 v 1.2
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x
Marlin 2.0.9.1

### Additional Information

[ConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7891433/ConfigFiles.zip)

oldman4U, 2022-01-18:
Does it work using Marlin?

seth995, 2022-01-18:
> 

Yes, via Pronterface g26 executes with no problem

digant73, 2022-01-19:
@seth995 it's a bug in Marlin. In my case the mainboard freezes while the TFT is still working. Same result if I send G26 from Pronterface. Tested on latest (2 days ago) official 2.0.9.3 Marlin release. Open a ticket in Marlin repo.

oldman4U, 2022-01-19:
I use 2.0.9.1 and it works.

oldman4U, 2022-02-01:
What is the status of the ticket?

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2022-02-04:
@Msq001 

Please close this ticket. Thank you

closed 2022-02-08 by bigtreetech