bigtreetech/BIGTREETECH-TouchScreenFirmware #2330: [Q] Wrong type used in PrintingMenu.c switch case?

rondlh, 2022-01-18:
I just noticed something odd in the source code in PrintingMenu.c, line 752.
```
    KEY_VALUES key_num = menuKeyGetValue();
    switch (key_num)
    {
      case PS_KEY_0:
        heatSetCurrentIndex(currentTool);
        OPEN_MENU(menuHeat);
        break;
```
**PS_KEY_0** is not part of the enum type **KEY_VALUES**, it's part of **PS_KEY_VALUES**

So shouldn't it be?
`PS_KEY_VALUES key_num = menuKeyGetValue();`

I also noticed that KEY_LABEL_0 - KEY_LABEL_7 of KEY_VALUES are never used, there is no action linked to it anywhere, They are the labels underneath the icons, which when touching result in audio touch feedback, but never result in any action. Wouldn't it make more sense to make the ICON + label to be one area, which results in an action when either is clicked?

digant73, 2022-01-20:
In theory (wrong casting) your advice is correct but menuKeyGetValue internally handles different kind of scenarios. One of that is when in PrintingMenu. So the values provided in the switch statement are the correct ones

rondlh, 2022-01-20:
OK, understood, thanks for your reply. For a quick test I updated the code so that pressing the labels acts the same as pressing the ICON. I always found it annoying that when I look at the printer and press an ICON, often I hear the press, but nothing happens because I pressed the label accidentally. If there would be no sound then that would be fine too, but a press sound without a response feels odd.

closed 2022-01-21 by rondlh