bigtreetech/BIGTREETECH-TouchScreenFirmware #2337: [BUG] Portrait Mode print summary broken

radry, 2022-01-20:
The Print Summary screen when using Portrait Mode is broken.

I think it's this part in PrintingMenu.c that causes it:
Line 492 ff

```
void drawPrintInfo(void)
{
  GUI_SetTextMode(GUI_TEXTMODE_TRANS);

  IMAGE_ReadDisplay(rect_of_keySS[KEY_INFOBOX].x0, rect_of_keySS[KEY_INFOBOX].y0, INFOBOX_ADDR);

  GUI_SetColor(INFOMSG_BG_COLOR);
  GUI_DispString(rect_of_keySS[KEY_INFOBOX].x0 + STATUS_MSG_ICON_XOFFSET, rect_of_keySS[KEY_INFOBOX].y0 + STATUS_MSG_ICON_YOFFSET,
                 IconCharSelect(CHARICON_INFO));
  GUI_DispStringInRectEOL(rect_of_keySS[KEY_INFOBOX].x0 + BYTE_HEIGHT + STATUS_MSG_TITLE_XOFFSET,
                          rect_of_keySS[KEY_INFOBOX].y0 + STATUS_MSG_ICON_YOFFSET,
                          rect_of_keySS[KEY_INFOBOX].x1 - STATUS_MSG_TITLE_XOFFSET,
                          rect_of_keySS[KEY_INFOBOX].y1 - STATUS_MSG_ICON_YOFFSET,
                          (uint8_t *)textSelect(LABEL_PRINT_FINISHED));

  GUI_SetColor(INFOMSG_FONT_COLOR);
  GUI_SetBkColor(INFOMSG_BG_COLOR);
  GUI_DispStringInPrect(&msgRect, LABEL_CLICK_FOR_MORE);
  GUI_RestoreColorDefault();
}
```

The problem is it uses the same box as the status screen. But this doesn't work with the portrait printing menu.
I would fix it myself but I don't have a good solution how it should look like, since it's a unique Layout for portrait mode.

### Steps to reproduce

1. Be in Portrait Mode
2. Finish a print

**Actual behavior**
[image]

### Hardware Variant
TFT24
### TFT Firmware Version & Main Board Firmware details
Current master

digant73, 2022-01-21:
assuming there is no space enough to display the box on the right of the fan icon, it seems the box must be suppressed

radry, 2022-01-21:
@digant73 
The info box is as wide as two normal icons. As you can see compared to the bottom row, there isn't enough space next to the fan icon for that.

digant73, 2022-01-21:
yes I see. Unfortunately it seems the only viable option is to remove that info

radek8, 2022-01-22:
If the "Main" icon moves up over the "Back" icon, there is enough space in the information field to the left of the "Back" icon

digant73, 2022-01-23:
@radek8 yes thanks. that should be the better choice. Currently I'm focused to fix some other issues. If this issue is not fixed by someone else I will see it in the future

radry, 2022-01-23:
@radek8 
Yes but it would require a separate "layout" just for this one screen and a lot of code rewrite. I don't see much benefit in this infobox anyway. I think it would be easier to call a "popup" window that says "print finished".

PR #2345 (2022-01-25): Bugfixes in Terminal + cleanup
**BUGFIXES:**
* Fixed some bugs in Terminal: Terminal had some general bugs such as wrong rolling pages, wrong page count, wrong colors (e.g. orange for ACK messages), the prev button increased the page (instead of decreasing it), next button decreased the page (instead of increasing it), when the buffer was rolled the page count was reset to 1, TFT freeze (infinite loop) if character '\0' was present in the buffer etc. It fixes also #2304.
* Minor fix on LEDColor menu: LED color was not changed when increasing/decreasing an RGB component not having the focus. Al [... cut]
closed 2022-01-25 by bigtreetech