bigtreetech/BIGTREETECH-TouchScreenFirmware #2339: M109 in "start_gcode" does not wait for temp reach

masintxi, 2022-01-21:
I'm trying to purge the hotend before start printing anything (I usually do this in the start gcode of the slicer). I've tried to add the same gcode sequence but it gives me a "cold extrusion" warning

I'm using a BTT TFT35 v3.0 (with the latest firmware available) and a BTT SKR mini E3 v2.0 (with Marlin bugfix 2.0.x)

This is the gcode I use in the slicer and it's working properly:
```
G28 R15                  
M82                   
G0 X0 Y200
M107
M109 S200                          
G92 E0                
G1 E20 F200          
G92 E0
```

And this is the sequence I'm using in the "start_gcode" variable:
`start_gcode: G28R15\n M82\n G0X0Y200\n M107\n M109S200\n G92E0\n G1E20F300\n G92E0\n`

The rutine works because the extruder moves to the position I want (X0 Y200) and the hotend starts heating. The problem is that it inmediatly tries to extrude but the printer is still cold so the "cold extrusion" warnings pops up.

I've not found anything about issues with the M109. Is this a limitation of the TFT or am I doing something wrong?

Thanks in advance!

rondlh, 2022-01-21:
> You should separate "M109S200" as "M109 S200" and disable "Emulated M109 And M190" in "config.ini" (emulated_m109_m190:0). That should do it.

Why would you  need to separate "M109S200"? Gcode does not care about spaces, except for comments and such.

kisslorand, 2022-01-21:
Readability?
Anyway, you're welcome.

masintxi, 2022-01-21:
Thank you so much for the quick reply @kisslorand .  I completely forgot about it, I'll try that first time tomorrow.

I enabled the setting because I'd like to see how the temp goes up and I understood that it would not happen with the setting disabled (_"the communication (TFT<->Marlin) will be frozen until desired/set temperatures are obtained"_).  Anyway, I'll try this and check the behavior.

Thanks again!

masintxi, 2022-01-22:
I've tested disabling the "Emulated M109 And M190" setting and it now works perfectly. Thank you @kisslorand 

That said, I still can see the hotend temperature readings while waiting. If the communications with the board fw were really frozen I shouldn't be able to get the temps. What does this setting affect to?

kisslorand, 2022-01-22:
The mainboard is not taking any commands but it is set to auto report temperatures.

masintxi, 2022-01-22:
Understood. I don't think I have any reason to enable the setting.

I proceed to close the issue

Thanks again for the help

closed 2022-01-22 by masintxi