bigtreetech/BIGTREETECH-TouchScreenFirmware #2366: Bug change filament

valerox84, 2022-01-30:
### Description

you heat to make the change of filament, you give it to load and does not give you option to download filament, all the time gives the user to paused

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2022-01-30:
In Marlin, enable all the settings reported at the beginning of TFT's config.ini file

valerox84, 2022-01-30:
If I have everything enabled in configuration.adv, instead in marlin mode it does it perfectly, instead in tft mode that error happens.
my board is skr e3 dip

digant73, 2022-01-30:
What's the version of Marlin?

valerox84, 2022-01-30:
version 2.0.7.2 btt e3 dip,I always used the marlin mode version, but lately the screen gets strange symbols, instead in tft mode it works fine, but the change of filament is what does not work.

radek8, 2022-01-30:
Version 2.0.7.2 is more than a year old.
Update to new versions.
Attach the Marlin configuration files.

digant73, 2022-01-30:
The minimum Marlin version required by the TFT fw is 2.0.8.1. I suggest to move to latest Marlin fw (e.g. 2.0.9.3 or even 2.0.9.1)

oldman4U, 2022-02-01:
@valerox84

Usually the owner of a ticket answers so that those trying to help know what's going on. 

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2022-02-03:
@Msq001 Please close this ticket.

Thank you

closed 2022-02-03 by valerox84