bigtreetech/BIGTREETECH-TouchScreenFirmware #2368: INVERTED_Z_AXIS  1 not work

lnfactionfreddy, 2022-01-31:
### Description
The Define INVERTED_Z_AXIS  1 not work at move menu on this display on touch mode.
it has no effect on the movement correctly.

### Hardware Variant
TFT 70 V3.0

digant73, 2022-01-31:
If you used config.ini, all the hard coded settings in Configuration.h are replaced with the settings configured in config.ini. So if you did it, in config.ini verify and change the settings in "inverted_axis" according to your need (as done in Configuration.h)

rondlh, 2022-02-01:
I can confirm that the inverted Z axis works if configured in Config.ini (inverted_axis:X0 Y0 Z1). 
The icons, labels and move directions are changed as expected. Tested on BTT TFT35 V3.0.

QuestionDevelopment, 2022-02-02:
I can confirm that inverted Z axis configuration does not worked.  Tried running both settings for my z axis and no change at all.

digant73, 2022-02-02:
@QuestionDevelopment Please, pay attention to what I already reported above and verify it matches with your need. Also, close your ticket. It is a duplication of this one

QuestionDevelopment, 2022-02-02:
@digant73 : Ticket closed.  Please ignore my report.  At this time there are too many confounding variables for me confidently state if the config setting was or was not working.  All I feel confident saying at this time is that setting the value in Configuration.h and then restoring default values allows the inverted setting to be utilized.  Thank you for your help.

lnfactionfreddy, 2022-02-02:
the configuration of the config.ini solved the problem sorry for opening the ticket

closed 2022-02-02 by lnfactionfreddy