bigtreetech/BIGTREETECH-TouchScreenFirmware #2369: TFT35 E3 V3.0 PID calibration

Grzesinski, 2022-02-01:
### Description

In the TFT35 E3 V3.0 display (with SKR MINI E3 V2.0) PID calibration doesn't work properly.
### Steps to reproduce

1. Choose PID calibration
2. Choose Bed
3. Start calibration
4. Check is the bed warming
5. Wait for final message

**Expected behavior**
Hotend or bed PID proper calibration.

**Actual behavior**
Regardless of the choice, the head is always heated and the process always ends with an error.
Calibrating directly from the TFT terminal using M commands working properly.

### Hardware Variant
SKR MINI E3 V2.0 (Marlin 2.0.9.1)
TFT35 E3 V3.0 (BIGTREE_TFT35_V3.0.27.x.bin)

digant73, 2022-02-01:
TFT fw for E3 is BIGTREE_TFT35_V3.0_E3.27.x.bin (you reported BIGTREE_TFT35_V3.0.27.x.bin maybe by mistake).
Furthermore, PID menu sends the following gcode for bed:
`M303 U1 C8 E-1 S<bed temp>`
where <bed_temp> is the selected target temperature (e.g. 60).
Try to send that command from terminal (or pronterface etc...) and see if it works.

radek8, 2022-02-01:
Why are you opening duplicate tickets? # 2335
On my printer, if I run the bed PID, the bed will start to heat up.

Grzesinski, 2022-02-02:
> TFT fw for E3 is BIGTREE_TFT35_V3.0_E3.27.x.bin (you reported BIGTREE_TFT35_V3.0.27.x.bin maybe by mistake). Furthermore, PID menu sends the following gcode for bed: `M303 U1 C8 E-1 S<bed temp>` [...]

I checked my software versions again.  My mistake.
SKR MINI E3 V2.0 Marlin 2.0.9.1 12.08.2021)
TFT35 E3 V3.0 (TFT35_E3_V3.0.27.x 6.10.2021)
I will check command from TFT terminal today (E0 also).
So far I used the M303 H1 (or H0) commands and it worked (it was heating the right heater).

digant73, 2022-02-02:
option H is not even reported as a valid option for M303 (see https://marlinfw.org/docs/gcode/M303.html).
If the gcode, I reported previously, will not work, try to update to Marlin 2.0.9.3 (it is working as expected)

Grzesinski, 2022-02-02:
OK.
But then requirement information "Minimum Marlin firmware version: 2.0.8.1" should be updated also.
I try test all today.

Grzesinski, 2022-02-03:
M303 U1 C8 E0 S180 works, hotend is heated.
But then I leave the TFT terminal, an additional message "PID autotune aborted" appears. 
Even if I wait an hour and save the E0 PID settings using M500. Strange.

M303 U1 C8 E-1 S60 (or E1) stopped with error "PID Autotune failed! Bad heater id." 

Now I'm configuring & compiling Marlin 2.0.9.3.

radek8, 2022-02-03:
Do you have PID beds in Marlin enabled?

radek8, 2022-02-03:
M303 U1 C8 E-1 S60
Why use the U1 switch?

digant73, 2022-02-03:
@radek yes, it was used to apply the new PID values so with M500 they are also made persistent
@Grzesinski forget the error message "PID autotune aborted" you got (it is correct if you didn't use the PID menu).
For the bed issue, as suggested by Radek, verify PIDTEMPBED is enabled in Marlin's Configuration.h file

Grzesinski, 2022-02-03:
You're right. I have 
//#define PIDTEMPBED in Configuration.h in current Marlin 2.0.9. 1
This is default BTT configuration for SKR MINI E3 V2.0 and Ender 3 Pro (for Marlin 2.0.9.1 and 2.0.9.3.)
This file contains also comment "FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles."

Does it mean that the table was bang-bang controlled ? 

I will test this setting.

Grzesinski, 2022-02-03:
OK.
Now PID works properly with Marlin 2.0.9.1
It would be good to include this information in the TFT configuration requirements. 
Unfortunately I cannot test Marlin 2.0.9.3 due to compilation errors. 
Error occurs with default BTT config files. Even manually editing default Marlin configuration.h and configuration_adv.h not helped (error 'manual_feedrate_mm_s' was not declared in this scope; did you mean 'feedrate_mm_s'?)
Either way, we can assume that PID should will work.

radek8, 2022-02-04:
PID is not required for display operation.
Someone doesn't have to use it. 
If I don't have a heated bed, the bed turns on the relay, or the PUD of the bed causes interference, I won't turn on the PID.

radek8, 2022-02-04:
Be careful when transferring configurations between Marlin versions
If you are testing multiple versions of Marlin, you must use the configuration files for that version, which you will modify according to your configuration.
You must never copy an entire configuration file from one version to another.

digant73, 2022-02-04:
@Grzesinski if you solved your issue, please close this ticket

Grzesinski, 2022-02-04:
Thanks for support.
Of course bed PID isn't necessary but bang-bang steering may cause print layers artefacts.
I don't know why isn't default enabled in SKRmini configuration.
I will check new Marlin. Errors appears at default configuration files for SKR from current repository.

closed 2022-02-04 by Grzesinski