bigtreetech/BIGTREETECH-TouchScreenFirmware #2371: [FR] (feature request title)

shafr, 2022-02-01:
**Is your feature request related to a problem? Please describe.**
After latest update .29 my TFT24_v1.1 stopped working - no touch & white screen

**Describe the solution you'd like**
IT would be nice if there was an instruction how to do emergency recovery / flash using Serial adatper using PC & Platformio

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

radek8, 2022-02-01:
Did you use reset with the reset.txt file?

radek8, 2022-02-01:
What version of .29?
The current version is TFT24_V1.1.27

shafr, 2022-02-01:
@radek8 - sorry, yes version is .27.
Also - what is the difference between:
* BIGTREE_GD_TFT24_V1.1.27.x.bin
* BIGTREE_TFT24_V1.1.27.x.bin

What's GD? 

MY main concern now - that touch had stopped working with .27, like UI buttons do not react, but the physical touch is working (backlight dims & lights up again)

radek8, 2022-02-01:
GD is a different HW.
You must use the correct version.
perform a reset after the update.
Copy the empty reset.txt file to the card, paste it into the TFT and turn on the display.

shafr, 2022-02-01:
Copying `reset.txt` helped. Thank you.

closed 2022-02-01 by shafr