bigtreetech/BIGTREETECH-TouchScreenFirmware #2374: [BUG] Artillery Genius filament not working after update

Alpacad, 2022-02-02:
### Description
I'm working on my friend's Artillery Genius, after flashing the bigtreetech tft firmware the filamente runout sensor has stopped working. This was months ago and we simply decided to wait some bug fixes. After reading about #2007 and the fix #2018 we decided to retry.
But the problem is still there. The sensor is red during flashing, then turn green even if the filament is not inserted.

### Steps to reproduce

1. Download the latest firmware
2. Customized config.ini to match Artillery Genius values
3. Flash firmware

**Expected behavior**
Filament sensor should work without problems.

**Actual behavior**
Filament sensor led is constantly green and a print cannot be started unless deactivated from the menu.

### Hardware Variant
MKSTFT28
### TFT Firmware Version & Main Board Firmware details
MKS_TFT28_NEW_GENIUS.27.x.bin
Marlin 2.0.9.3 on MKS Gen L 1.0

### Additional Information
Attached files:
- config.ini
- Configuration.h

[GeniusFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7989682/GeniusFiles.zip)

kisslorand, 2022-02-02:
> Filament sensor led is constantly green...

Is it like this regardless of filament being present or not?

kisslorand, 2022-02-02:
Try with MKS_TFT28_V4.0.27.x.bin.

Alpacad, 2022-02-02:
Already tried, without success. Used the same config file and firmware version.

kisslorand, 2022-02-03:
The printer is a Genius Pro (with BlTouch like leveling sensor)?

Alpacad, 2022-02-03:
No, it's not mine, but I'm pretty sure it's not a Pro. No bltouch-like sensor or 32 bit mainboard. It has the  dedicated expansion port on the side and MKS Gen L 1.0 (atmega2560 based) board.

kisslorand, 2022-02-03:
The things you described is the result of the filament runout detection sensor being connected to other pins than the MCU on the TFT expects it.
I see two possible fixes:
1. make a FW patch to reflect the actual connection
2. reconnect the filament runout sensor to other pins on the TFT

I can help with either of then but I recommend to rewire the sensor so it will work with every future FW release.

I made two test files for you to try, they are meant to test if the sensor is connected to other pin than expected on the TFT.
These files are based on the master version so it's safe to use any of them.
Here's the link: https://mega.nz/folder/vthEHaBa#Ql5oXvKNm34Ji3mJrs3DpA

Alpacad, 2022-02-03:
I've passed the files to my friend. He said he'll try them this weekend, I'll let you know.

Alpacad, 2022-02-10:
Sorry, @kisslorand, my friend didn't try to flash the files yet. 
Please, accept my apologies. As I said, I'll let you know as soon as he texts me the results.

Alpacad, 2022-02-10:
Here we are. He says that none of the files worked for him, and the sensor remains green all the time.
Just to be sure, these .bin files had to be flashed alone or included with the other .ini and folders?

kisslorand, 2022-02-10:
They not need newer or other files that are already used for the master branch. So, yes, it's enough to flash the bin files alone.
As for the persistence of the issue... I would need a picture of the TFT's back side as clear as possible in the region where the filament sensor is connected. 

Something jus caught my eyes... The LED is constantly green? Isn't green supposed to mean that there IS filament? Yet the TFT sees trigger for no filament?

oldman4U, 2022-02-16:
What is the status of this ticket?

Please help the community and close the ticket once you do not need it anymore. Thank you

Alpacad, 2022-02-17:
I'm still waiting a response, my friend is a bit lazy when it comes to test something or fix problems. I have to remind him daily.
It's funny because yesterday he was printing holders and asking me for a custom firmware with bltouch and auto bed levelling.

kisslorand, 2022-02-17:
> I'm still waiting a response, my friend is a bit lazy when it comes to test something or fix problems.

Than I guess it will be no problem if we'll get lazy fixing his problems...

Alpacad, 2022-02-21:
These are the photos that he sent to me. As for the parameter, he said that it does not let him print.

[image]
[image]
[image]

kisslorand, 2022-02-21:
From the pictures it is a newer Genius it must work with NEW_GENIUS FW. Maybe it worth a try and move the filament sensor connector to the next row of pins (B0) and try with regular Genius FW.
As for the printer not working with neither settings for the  "fil_runout_inverted" parameter, something is fishy. It's just simply not possible. Something else also must happened, not just a simple FW update. 
What happens if the connector is unplugged from the sensor and  "fil_runout_inverted" is tested with both values (0 and 1)?

Alpacad, 2022-02-24:
@kisslorand we may have discovered a bad connection between the tft the jst-xh connector. I'll make him do some more tests and let you know.
If if that's it, we may consider rewiring the connector (maybe with a dupont one) and putting a drop of hot glue.

oldman4U, 2022-02-24:
So the title of this ticket is wrong and the ticket is not related to the update, or kisslorand's code is now even able to break cables;-)

May the force be with you

Alpacad, 2022-02-24:
Could be something my friend did while upating the firmware, like pulling the cable. So seeing this after the update, first thing was thinking about a possible bug.

Anyway, it seems that now it changes color to green when the filament is inserted and red when removed. But it doesn't seem to stop when trimming it. We'll do some other tests.

Alpacad, 2022-02-25:
Update:
Filament sensor color works as expetced, but it doesn't trigger filament runout script. If fil_runout_inverted is set to 1, it triggers immediatly with or without filament.

kisslorand, 2022-02-25:
Where is it the sensor connected now (B0 or B1) and what FW was used?

kisslorand, 2022-03-23:
@bigtreetech 
I think this can be closed. The user (or his friend) is not interested in resolving his own issue.

oldman4U, 2022-03-28:
@Msq001

Please close this ticket

Thank you

Alpacad, 2022-03-28:
Closing due to lack of interest by the user.

closed 2022-03-28 by Alpacad

jamarju, 2022-12-21:
I had this same problem. The fix was setting the sensor pin to B0 instead of B1. On `pin_MKS_TFT32_V1_4.h` replace this:

```C
    #define FIL_RUNOUT_PIN PB1
```

with this:

```C
    #define FIL_RUNOUT_PIN PB0
```

**That's in spite of the connector being connected to B1 according to the board's silkscreen**. See photos below. Perhaps the PCB silkscreen is wrong, in retrospect I should have traced the header back to the actual MCU pin but now I'm too lazy to take the printer apart. Anyway, B0 works, yay!

[image]
[image]
[image]
[image]