bigtreetech/BIGTREETECH-TouchScreenFirmware #2378: [BUG] failed to enable bed leveling after setting deafult

jnslnk, 2022-02-03:
Hey there,
after i set my Printer settings to deafult the display shows: "failed to enable bed leveling"
I've got a BIQU B1 (SKR V1.3) with no probe, so i don't under stand why it's trying to ABL anyway ?!
Anything i can do in Marlin Firmware or by sending the Printer some G-Code, to tell it it doesn't need to probe?
Thx 4 helping!

radek8, 2022-02-03:
You probably have some kind of bed leveling set up in the Marlin configuration.
Please correct the Marlin configuration and you will not receive an error.

radek8, 2022-02-03:
If you do not want to level the bed, you must comment on all these parameters

_//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING_

radek8, 2022-02-03:
Set the parameter in the TFT configuration file _config.ini_
auto_load_leveling: 0

oldman4U, 2022-02-11:
@Msq001 

Please close this ticket.

Thank you

closed 2022-02-15 by bigtreetech