bigtreetech/BIGTREETECH-TouchScreenFirmware #2379: [BUG] TFT35 V3 touch function does not work!

AcidSubway, 2022-02-03:
I have equipped my Artillery SWX1 with a new display TFT35 V3. After installation, the display has shown everything. Without touching it I thought I'd do a firmware update to BIGTREE_TFT35_V3.0.27.x.bin. The update was successful and the display booted up again. At the moment when I should calibrate the display I had a power failure. Now I restarted the device and found that the touch operation does not work. Is it because I could not run this during the update? I have completed the update again but there is no more display calibration. What to do, please for your help. 

best.g.

radek8, 2022-02-03:
Copy the empty _reset.txt_ file to the SD card.
Insert the card into the TFT and reset the TFT.

radek8, 2022-02-03:
Read the instructions on the home page before you open the ticket.
Troubleshooting section

AcidSubway, 2022-02-03:
SORRY redek8,

its worked MEGA THX!!!

AcidSubway, 2022-02-03:
Simple Reset there is ;/ SORRY

digant73, 2022-02-04:
@AcidSubway please close the ticket if Radek's suggestions solved your issues

closed 2022-02-04 by AcidSubway