bigtreetech/BIGTREETECH-TouchScreenFirmware #2381: "ok" response not sent to supplementary serial ports, in certain cases[BUG] (short description)

pillopaolo, 2022-02-05:
I just moved from 0.26 to 0.27 and noted that most likely there is a bug in parseAck.c
line 1242 --> ack_port_index = PORT_1

Due to this line, when I send a gcode from a different application (e.g Pronterface, 3DFox Android app, etc.) connected via UART4 (in my case BlueTooth module), the response is truncated.
In particular, only the first response line is send back to the supplementary serial port which originated the gcode command.
As a consequence, the second response line, usually the "ok" line, is missing.

This is not good: some applications, and the user him/herself, expects that "ok", to be sure that everything went well.

Expected behaviour:
M220
   FR:100%
   ok
M221
   echo:E0 Flow: 100%
   ok
M114
   X:-3.00 Y:-8.00 Z:0.02 E:0.00 Count X:-240 Y:-640 Z:4
   ok

Current (wrong) behaviour:
M220
   FR:100%
M221
   echo:E0 Flow: 100%
M114
   X:-3.00 Y:-8.00 Z:0.02 E:0.00 Count X:-240 Y:-640 Z:4

I am sure that code line was meant to do something useful as well, however is causing this problem.
Can you pls fix it?

Thanks

digant73, 2022-02-05:
yes, that's a bug. Using "if (infoHost.wait==false) ack_port_index = PORT_1;" is it working as expected right?

pillopaolo, 2022-02-05:
@digant73 It works for me, but I am not sure 100% that (infoHost.wait==false) is the right/best way to capture the end of the multi-line response. You are the expert!

digant73, 2022-02-05:
yes, that condition is correct

kisslorand, 2022-02-05:
@pillopaolo 
Why do you need the "ok" response? Getting a response like "FR:100%" already means everything went well. The "ok" response means something totally different.

pillopaolo, 2022-02-05:
> @pillopaolo
> 
> > some applications, and the user him/herself, expects that "ok", to be sure that everything went well
> 
> Why do you need the "ok" response? Getting a response like "FR:100%" already means everything went well. The "ok" response means something totally different.

TFT should be as transparent as possible in forwarding Marlin responses to other programs that might be listening on other serial ports. Other programs (e.g. 3DFox) uses Marlin standard (i.e. "ok" at the end of each response) to detect if a multi-line response was completed.

After all, TFT itself uses Marlin "ok" to learn when a response is completed (look at parseAck.c code and search for "ok", you see how it is used).

kisslorand, 2022-02-05:
I know very well what the response "ok" means, it can be seen in the Marlin's source code. I wasn't questioning the need of receiving "ok", it was just a simple question asking what do YOU need the "ok" for. What is it that is not working for you because of that?

pillopaolo, 2022-02-06:
> I know very well what the response "ok" means, it can be seen in the Marlin's source code. I wasn't questioning the need of receiving "ok", it was just a simple question asking what do YOU need the "ok" for. What is it that is not working for you because of that?
> 
> I am asking you BECAUSE I know what "ok" is for.

I am the developer of 3DFox for Android. The app relies on "ok"s and other standard Marlin responses for managing the queue of command sent and responses, exactly as TFT does. And this bug messes up things.
I do not use Octoprint, maybe it has an influence of that too.

kisslorand, 2022-02-06:
OK, that's the info I was after. You need 'ok" for flow control.

mlaci001, 2022-02-07:
@digant73 
Hello,
Thank you for your work.
I tried to translate to BIGTREE_TFT35_E3_V3_0,
but I got the following error:
TFT \ src \ User \ API \ Vfs \ Source \ usbh_msc_bbb.c: 35:10: fatal error: usbh_pipe.h: No such file or directory.
Can you help
Thank

digant73, 2022-02-07:
@mlaci001 Simply retry to compile. It seems one of the minor bugs present in vscode.

mlaci001, 2022-02-08:
@digant73 Thanks, that was really the solution, but I had to download it and add the source to vsc again. Who understand this ?

closed 2022-02-15 by bigtreetech