bigtreetech/BIGTREETECH-TouchScreenFirmware #2382: TFT 24 doesn't work

Aleks1967, 2022-02-06:
Hello all, a question, I bought a tft 24 v1. connect to the board skr 1.4 turbo. firmware 1.27 in marlin mode on the screen says - marlin mode and nothing else, if I connect all 3 cables, the screen is lit white, and does not respond to any action, in touch screen mode, the picture is, when you touch icons do not happen, tell me what the cause of not working display

Aleks1967, 2022-02-06:
[image]
[image]
[image]
[image]

oldman4U, 2022-02-06:
Read the Troubleshooting section of the ReadMe on the front page of this repository. 

Please help the community and close the ticket once you do not need it anymore. Thank you

Aleks1967, 2022-02-06:
дайте пожалуйста ссылку на статью

Aleks1967, 2022-02-06:
Please give me a link to the article

kisslorand, 2022-02-06:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Aleks1967, 2022-02-06:
[image]

Aleks1967, 2022-02-06:
I do not understand what I need to read to fix, and the device began to work properly, I write with a translator, show me exactly where it is written about this error, thank you in advance
in vscode the line is uncommented as written in the instructions, what else needs to be done

radek8, 2022-02-06:
Send the configuration files and photos involved.

Aleks1967, 2022-02-06:
[Configuration h.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8010170/Configuration.h.txt)
[Configuration adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8010171/Configuration.adv.txt)

Aleks1967, 2022-02-06:
I sent configuration files with txt extension, because the original ones are not accepted by the site

Aleks1967, 2022-02-06:
What should be in the photos?

Aleks1967, 2022-02-06:
here is a photo of the pinout of the skr 1.4 turbo board pin TXO is in the middle of the pad, and in the TFT pad 24, it is 2 (UTX). in the cable the wires are not shifted relative to the location of the pins
I have changed the location of the wires, it did not change anything, the screen when you connect the 5 pin wire is white, disconnect the wire and the display begins to show the picture,
In Configuration adv, I did not change the lines for the touch screen, I need it only in Marlin mode

radek8, 2022-02-06:
Send a photo to see the connections and orientations of the EXP1 EXP1 and TFT tables on both sides.

radek8, 2022-02-06:
Uncomment
#define M114_DETAIL
#define REPORT_FAN_CHANGE
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define EMERGENCY_PARSER

In Marlin you have a speed of 250000
In config.ini set
serial_port: P1: 7 P2: 7 P3: 0 P4: 0

radek8, 2022-02-06:
Update the display to the latest FW.
If the TFT does not respond to the touch, perform a reset with the reset.txt file.
Then upload the configuration with the config.ini file.

Send those photos. It looks like a badly connected cable

Aleks1967, 2022-02-06:
[image]
[image]
[image]
[image]
[image]

Aleks1967, 2022-02-06:
the marlin version installed for the display is the latest
In config.ini set
serial_port: P1: 7 P2: 7 P3: 0 P4: 0 - What needs to be fixed? Is it the display config?

Aleks1967, 2022-02-06:
[image]

Aleks1967, 2022-02-06:
should be corrected here? it should be set to -6, which corresponds to 250000
Do I understand it correctly?

kisslorand, 2022-02-06:
No, 6 corresponds to 115200 ("115200: 6").
For 250000 select 7 ("250000: 7").
As you can see in the config.ini:
 - OFF (port disabled): 0
 - 2400: 1
 - 9600: 2
 - 19200: 3
 - 38400: 4
 - 57600: 5
 - 115200: 6
 - 250000: 7
 - 500000: 8
 - 1000000: 9

radek8, 2022-02-06:
Check the serial cable (black 5 pins). Take a picture of the color markings on the cable.

radek8, 2022-02-06:
The cable seems to me to be rotatable by 180 degrees

Aleks1967, 2022-02-07:
[image]
I changed the data in the config.ini file, reflashed, nothing has changed, turned it 180 degrees EXP2 no result

Aleks1967, 2022-02-07:
[image]
Here is a 5 PIN cable

kisslorand, 2022-02-07:
You are changing the comment, the explanation itself, not the parameter
You should have this:  **serial_port:P1:7 P2:0 P3:0 P4:0**

Aleks1967, 2022-02-07:
with a 5 pin cable figured out, not properly connected to the motherboard, although the video on YouTube single cone connected to the right side, white screen ceased to be, reset using reset.txt file touch settings, the touch screen on the display began to respond to click and open tabs, Marlin mode does not work, before the inscription on the display was yellow letters, now whit
[image]
e

radek8, 2022-02-07:
Yes, it's as I expected.
You have a 180 degree rotation of the serial cable.

[image]

radek8, 2022-02-07:
Also check cables EXP1 and EXP2.
Orientation is correct.
But make sure you don't cross them.

Aleks1967, 2022-02-07:
made, here is the picture, no change, the inscription marlin mode has become red
[image]
The touch screen now works properly.
marlin mode does not work,

radek8, 2022-02-07:
Your problem is that you connect everything according to some instructions on Youtube and when it doesn't work, you try to turn the connectors headless.
This approach will destroy something.
Try to engage reason. download the pins and connect the cables of the same pins on the SKR to the same pins on the TFT.

[image]

radek8, 2022-02-07:
>  I changed the data in the config.ini file, reflashed, nothing has changed, turned it 180 degrees EXP2 no result

You must not rotate the EXP2 connector
He was well connected in the first photos.
Man, do what we advise you to do. don't make more mistakes.

kisslorand, 2022-02-07:
I am speechless... I cannot compete with this!

Aleks1967, 2022-02-07:
[image]
I flashed the display with this file, the marlin mode does not work

Aleks1967, 2022-02-07:
flashed, flashed, rebooted the printer, marlin mode does not work

Aleks1967, 2022-02-07:
I flashed, rebooted the printer, the marlin mode does not work

radek8, 2022-02-07:
Send the current configuration that you last uploaded to SKR and TFT
send photos of current connection cables.

Aleks1967, 2022-02-07:
[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8014460/config.txt)
The skr 1.4 configurations are the same as above, the stubs are as they should be, the firmware for the display is 1.27,

Aleks1967, 2022-02-07:
[image]
[image]
[image]

Aleks1967, 2022-02-07:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8015403/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8015404/Configuration_adv.txt)
here is the configuration with the touch screen display, I had to uncomment the lines regarding the 2 lines of the Z-probe. since I do not have a bl-touch sensor

Aleks1967, 2022-02-07:
I upgraded the skr 1.4 motherboard. the touchscreen display says connection - no, 
Marlin mode - does not work configuration files posted above, tell me what I did wrong, thank you in advance

radek8, 2022-02-07:
Do you have cables EXP 1 and EXP2 connected according to the picture?

[image]

Aleks1967, 2022-02-07:
Yes

radek8, 2022-02-07:
You did not send the configuration file config.ini.
In the SKR board is Marlin in the TFT board is the FW of the touch screen. These are two different SWs that must be configured.
Do you know how to update both?

Aleks1967, 2022-02-07:
[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8015631/config.txt)
Here is the configuration for the display

Aleks1967, 2022-02-07:
updated everything, I print through the slicer, I can not connect the display for half a year, this board with a display did not work, the old board burned out along with the display, skr bought earlier, then bought a display, thought it had no firmware and reflashed it immediately, I could not connect, well, looking for options from whom to ask, started asking questions here

radek8, 2022-02-07:
If you had a display connected to the SKR that burned out, it is quite possible that the inputs on the EXP1 and EXP2 connectors of the SKR board also burned out.

radek8, 2022-02-07:
if you upload the configurations you last sent, the display can work in both modes. If the SKR board has the inputs fired, Marlin mode will not run.

radek8, 2022-02-07:
Open on the touch screen.
Menu
Settings
Sport
See what speed you have set to 1. Printer.

Aleks1967, 2022-02-07:
the parameter is set correctly 250000, I click on connect, the inscription "connection" appears if I click on the button writes TFT listening mode and nothing work
[image]
s

radek8, 2022-02-07:
Noo
S.Ports

[image]

Aleks1967, 2022-02-07:
[image]
Maybe this setting needs to be uncommented

Aleks1967, 2022-02-07:
There the correct speed is 250000

radek8, 2022-02-07:
No changing the status screen does not solve anything.

Try setting different speeds and restart the display in the meantime.
He catches the one in Marlin

Aleks1967, 2022-02-07:
[image]
If I press this button, it says - your printer is controlled by the computer, usb cable is disconnected

Aleks1967, 2022-02-07:
I'm sorry, it's 1 a.m., can I have another consultation tomorrow?

radek8, 2022-02-07:
Don't click on anything else.

radek8, 2022-02-07:
OK 
I have 16:00 :-)

Aleks1967, 2022-02-07:
[image]
Checked like this, all 5 wires are ringing

Aleks1967, 2022-02-07:
[image]
I am alarmed by this pinout, TXO pin 2 on the display and 3 on the skr

Aleks1967, 2022-02-07:
On the 5 pin wire, no overlap

[57 more replies]