bigtreetech/BIGTREETECH-TouchScreenFirmware #2384: [BUG] Display on wrong resolution

nicedevil007, 2022-02-07:
### Description

### Steps to reproduce

1. Install KlipperScreen on the Pi
2. Adjust the /boot/config.txt on the pi
3. Turn the printer on

**Expected behavior**

See the KlipperScreen at fullscreen on the display.

**Actual behavior**

Screen looks like this :(
[image]

### Hardware Variant

BTT TFT7 => that one that is default one for the BiquBX

### TFT Firmware Version & Main Board Firmware details

The stock one from BTT on delivery of there Kickstarter Campaign... don't know where I can read it... long press on the button doesn't show any menu to switch.

### Additional Information
/boot/config.txt
```
max_usb_current=1
config_hdmi_boost=7
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60
hdmi_drive=1
```

Reference to this here => https://github.com/jordanruthe/KlipperScreen/issues/491

rondlh, 2022-02-07:
I don't think the firmware you run on the TFT is supported here.

radek8, 2022-02-07:
You are using your own FW. No one will help you with that.
Ask the author FW for help.
Your FW is portrait. Try rotating the display 90 degrees in the configuration.
Close the ticket.

closed 2022-02-07 by nicedevil007