bigtreetech/BIGTREETECH-TouchScreenFirmware #2385: [FR] Manual Mesh Bed Leveling Screen X&Y Values and I&J (grid point number) Values Should be Shown on the Scree

sarvenn, 2022-02-08:
I want to see the X&Y coordinates while probing the bed, this will let me adjust the probing margin accordingly. I can optimaze the margin to probe as much as off the bed but if I don't know the probing point coordinates I can not decide how to tune the probing margin from the edges. Also that would be better to show the grid point numbers (e.g. for a 3x3 grid I:0 J:0 is the first point I:2 J:2 is the last point).
I don't mean the mesh grid view screen. That screen also have it but only I and J values of the point.

So my FR requests to solve this issue are:
-Add Point I&J values on the MBL probing screen
-Add Point X&Y values on the MBL probing screen
-Add Point X&Y values on the MBL grid point screen

[image]
[image]
[image]
[image]

digant73, 2022-02-08:
all of those information are not available during the leveling process and it's not even possible to make a query during the process

sarvenn, 2022-02-09:
> all of those information are not available during the leveling process and it's not even possible to make a query during the process

But at leveling process X and Y gantries go to a certain coordinate at each point, after setting the Z offset at a point and clicking to the next button, head and bed moves to the next point, so I suppose that there should be a messege/command about it including the X and Y coordinates. And how about M114 or M154? Don't they report the current positions?

digant73, 2022-02-09:
the movement and positioning is controlled by Marlin. Only z offset must be changed during the process

sarvenn, 2022-02-09:
> the movement and positioning is controlled by Marlin according to the margin set in Marlin. Only z offset must be changed during the process.

I know that marlin takes care of the sequence but how about reporting the X and Y positions? Doesn't marlin report the x and y coordinates?
Can't we request the position information from marlin and report it on the probing sequence screen at MBL?
I know that when I go to terminal when I power on my printer, constantly marlin is reporting the coordinates.

radek8, 2022-02-09:
I defined the scan points in Marlin at compile time and they are still the same.
Why do I need to know the current value of the X and Y coordinates when leveling?
In the extreme case, I will count them and write them on the paper next to the printer.
Then I'll keep an eye on them.

sarvenn, 2022-02-09:
> I defined the scan points in Marlin at compile time and they are still the same. Why do I need to know the current value of the X and Y coordinates when leveling? In the extreme case, I will count them and write them on the paper next to the printer. Then I'll keep an eye on them.

For probe margin fine tuning (distances from the edges) and probing the as much as bigger surface of the bed. On a ender 3 due to home offsets there is wasted areas on the bed that you can't use practically so I need to use the right side of the bed as much as I can. To do that I need to probe the bed near at the right hand side of the bed edge and move the head further then the bed size (due to bl touch probe and nozzle offset). Therefore I'm seeking for the coordinates to fine tune them. Unfortunately when I go "back" to status screen while probing the bed (to see the coordinates from the status screen) sequence is cancelled. Not like the corner leveling function.

radek8, 2022-02-09:
I don't know what version of leveling you use.
I use Bilinear 5x5 points
By moving my head, I discovered her physical ability to move.
I adjusted the location of the endpoints for the probe accordingly.
I use a little modified Ender3 like you.

#if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT   10  //PROBING_MARGIN
  #define PROBING_MARGIN_RIGHT  10  //PROBING_MARGIN
  #define PROBING_MARGIN_FRONT  10  //PROBING_MARGIN
  #define PROBING_MARGIN_BACK   10  //PROBING_MARGIN
#endif

In my case, the probe scans 10mm corners from all sides.

Settings are also available for UBL and others.

#if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 180
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20
#endif

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif

oldman4U, 2022-02-12:
What are you going to do with this ticket now that you know the needed information is not available?

sarvenn, 2022-02-12:
@digant73 Isn't "auto report position" or "get current position" working during probing? (M154 / M114)

sarvenn, 2022-02-13:
// Extra options for the M114 "Current Position" report
#define M114_DETAIL         // Use 'M114` for details to check planner calculations
//#define M114_REALTIME       // Real current position based on forward kinematics

sarvenn, 2022-03-07:
[image]

PR #2497 (2022-04-20): Added X and Y coordinates in MBL and Mesh Editor menus
**IMPROVEMENTS:**
* Added X and Y coordinates in MBL and Mesh Editor menus: It implements #2385.

resolves #2385

**PR STATE:** Ready for merge.

[image]

[image]
closed 2022-04-20 by bigtreetech