bigtreetech/BIGTREETECH-TouchScreenFirmware #2386: Bigtreetech TFT43 Install fehler

boehm23, 2022-02-09:
Hallo,
 Ich bekomme keine Updates auf das Bigtreetech TFT43 V3.0 installiert Er bleibt immer hängen 
Mainboard ist Das SKR CR6 
wäre über hilfe Dankbar 
### diese Dateien habe ich verwendet
(BIGTREE_TFT43_V3.0.27.x-(THEME_Unified Menu Material theme/TFT43),config,language_de

PS sry ich bin ein kompletter Neuling in diesem Bereich
vielen Dank
[image]
[image]
[image]

kisslorand, 2022-02-09:
If you would have taken the curtesy to read the rules you would have found out that the spoken language around here is English.
Please play by the rules.

oldman4U, 2022-02-10:
And please read the ReadMe and the text on the screen before starting a ticket here.

This is not a user group but the BTT TFT FIRMWARE REPOSITORY. 

Please help the community and close the ticket once you do not need it anymore. Thank you

rondlh, 2022-02-10:
The Icons and fonts need an update, copy the required files to your SD card, insert into TFT and restart the display... 
Please close this ticket asap.

closed 2022-02-10 by boehm23