bigtreetech/BIGTREETECH-TouchScreenFirmware #2387: [Q] FYSETC SD-WIFI

xhejtman, 2022-02-09:
Is there anyone that manage FYSETC SD-WIFI card to get working with TFT35? I have one and I got mostly sd card init failed. It seems there is an issue maybe with timing or something. Not sure it is quite TFT35 related. I managed to fix fw of SD-WIFI so that WebDav interface works but it works only if it is not inserted into TFT35 card slot. It appears that TFT blocks somehow the card, so ESP is unable to read anything from the real card. Any ideas?

oldman4U, 2022-02-10:
BTT does not work either: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1343 Error working with BTT SD Cloud after a print

xhejtman, 2022-02-10:
It is a bit pitty that BTT SD Cloud does not work with BTT TFT35 E3 V3.0 ;)

oldman4U, 2022-02-10:
It does not work at all. ;-(

oldman4U, 2022-02-12:
Upload over WIFI would be great. I guess such a TFT would cost a bit more. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2022-02-12 by xhejtman