bigtreetech/BIGTREETECH-TouchScreenFirmware #2388: [FR] Lockout mode

rondlh, 2022-02-10:
I believe it would be useful to have a lockout mode where you can lockout unauthorized access to the TFT. 
I have 2 use cases for it:
1. Lockout my 2 year old from randomly pressing the touchscreen (he homed the printer while a part was still on the build plate, and gave himself a big applause, the probe mount broke)
2. Lockout colleagues in the office from playing around with the printers

An implementation could be like this:
1. Press the (new) lockout ICON to go back to the status screen, but any further TFT access requires a 4 digit pin code. You can still see the status, but are not allowed to make any changes without entering the pin code first.
2. Request the pin code after startup to get access (should be optional).
3. Auto locking could also be optional, triggered by the LCD Idle timer (should be optional)

kisslorand, 2022-02-10:
It's amazing how people are more concerned about the safety of their prints than the safety of their child...

> 1. Lockout my 2 year old from randomly pressing the touchscreen

I would rather lockout the child from the entire printer. It has to be done by the parent, it cannot be done by the FW.

rondlh, 2022-02-10:
> It's amazing how people are more concerned about the safety of their prints than the safety of their child...
> 
> > 1. Lockout my 2 year old from randomly pressing the touchscreen
> 
> I would rather lockout the child from the entire printer. It has to be done by the parent, it cannot be done by the FW.

You got a point there, but practically this also would protects the child. The print is running during the night, and he plays with the printer in the morning. If he cannot make is move of heat, then he is safe. I do not have a separate room for the printer. I already have hacked the firmware to make this work, by disabling the back button in the console, you need to give the correct key to be able to back out of the menu.

oldman4U, 2022-02-12:
How would this protect the child?

Please help the community and close the ticket once you do not need it anymore. Thank you

rondlh, 2022-02-12:
> How would this protect the child?
If the child cannot make the printer operate/move/heat then that is of course a good thing. For me the main issue is in the morning when the printer is finished already.
Another application of a lockout is to keep "smart" colleagues out, they like to demonstrate the printers to visitors, sometimes they start print jobs as a demo but never stop the 18 hour print...
I will close the ticket soon, let's see if there are more people interested in this. And as I said, I have made a quick hack already to achieve this (lock them in into the console menu), the drawback is that I cannot see the status easily anymore.

oldman4U, 2022-02-12:
Good you found a solution for your needs. 👍🏻

kisslorand, 2022-02-12:
These are very particular cases, the owner of the issue must find solutions for this. 
Printer can be put higher so little children cannot reach it, Put a remote power switch (Sonoff, Tuya, etc) for the main power so colleagues cannot start it, etc...

digant73, 2022-02-19:
@rondlh this FR is already pinned in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1170. See #1896.

closed 2022-03-02 by rondlh