bigtreetech/BIGTREETECH-TouchScreenFirmware #2389: How to remove pause ability at all

grishien, 2022-02-10:
Hello.
I'm getting spontaneous putting on pause with popup "Paused for user. Continue?".
This happens on its own, I don't know the reason.
I actually don't want to mess with the reasons. I want machine to totally loose the ability to put itself on pause, be it by me or firmware or bug or whatever. So I want the TFT to lose the ability to put on pause but force it to continue upon pause request.

**specify the device variant related to your question.**
TFT70

Guilouz, 2022-02-10:
Related this this issue #PR2295

radek8, 2022-02-10:
I dare to be sarcastic.
Are you a "lucky" owner of SKR 2 or SKRmini E2 V3.0?

radek8, 2022-02-10:
Try this PR. It could help.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2365

Guilouz, 2022-02-10:
@radek8 Working with SKR2, PR #2365 fix issue for me.

radek8, 2022-02-10:
@Guilouz  Well, I know, I'm watching the posts.
The information was intended for @grishien

grishien, 2022-02-10:
Thanks a lot. I've read through that branch. I've made changes to TFT FW accordingly and will be watching the result.
PS. I've  seen this problem earlier on SKR PRO and SKR2 with TFT70.

oldman4U, 2022-02-15:
Problem solved?

Please help the community and close the ticket once you do not need it anymore. Thank you

grishien, 2022-02-15:
Actually yes.
I followed the advise to split the M220/M221 commands.
But I went further. I ditched M221 since I'm CNC, not printer. So without M221 being fired instantly after M220 their collisions ended and problem was solved.

closed 2022-02-15 by grishien