bigtreetech/BIGTREETECH-TouchScreenFirmware #2391: [BUG] New Artillery not flashing LCD

Help3D-Padova, 2022-02-10:
### Description

There some Artillery Genius PRO or X2 that are not able to flash the LCD firmware.

### Steps to reproduce

Try to flash as usual

**Actual behavior**

MKSTFT28.bin is not detected and the flash is not able to start

### Hardware Variant

It seems that the new microcontroller is based on GD32F305 but i have to verify it with some user. 

### TFT Firmware Version & Main Board Firmware details

The bootloader is V3.0.4

### Additional Information

https://github.com/artillery3d/genius-pro-tft-firmware this firmware can be flashed without issue

@digant73 i'm not able to reproduce this issue because i have an older lcd but i have some user that can try some beta firmware.

kisslorand, 2022-02-10:
Try with lower case.

kisslorand, 2022-02-11:
This MCU is not implemented yet in this repository. AFAIK there's no BTT TFT with this MCU.

digant73, 2022-02-11:
@Help3D-Padova I also don't have this new TFT. That GD32F305 is not currently supported

Help3D-Padova, 2022-02-11:
> @Help3D-Padova I also don't have this new TFT. That GD32F305 is not currently supported

@digant73 i know, they are shipped with the last unit manufactured in december

digant73, 2022-02-19:
I would convert the type of the ticket from BUG to FR. That MCU is currently not supported

Help3D-Padova, 2022-02-26:
@digant73 my fault, it is not a bug :) 

The MCU is not supported but how can i help to find some information about it ?

Help3D-Padova, 2022-02-27:
ok i'll find someone that can send me a spare LCD with that MCU

digant73, 2022-02-27:
Try to ask here https://github.com/darkspr1te/BIGTREETECH-TouchScreenFirmware.
darksp1te was the guy that made the porting for MKSTFT35.

reynol3232, 2022-03-12:
I too have the GD32F305 chip set in the TFT of the Genius Pro.  I received my printer 3/11/2022.  I would like to find a way to resolve this issue and upload the more functional Marlin firmware.  The current installed bootload is 3.0.4 and does not read the micro sd upon booting.  I also have an older Artillery X1 and have updated that printer to Marlin.  The GUI for the new Genius Pro is very different and has less functionality.   Please help!
[image]

Help3D-Padova, 2022-03-14:
@reynol3232 flashing Marlin is not a problem because GD32F305 is related to the MCU of the LCD screen. Marlin i s uploaded on the Artillery Ruby

PrEzi3, 2022-03-14:
Exactly as Help3D said. Mainboard firmware is not a problem. The TFT MCU is. Currently I am also waiting for the GD32* variant to be supported in order to publish a "combo" of firwares for SWX2/GP including the new TFTs
https://www.thingiverse.com/thing:5186707

reynol3232, 2022-03-15:
I currently am not able to use my machine.  I loaded the files onto a microsd: config.ini, firmware.bin, MKSTFT28.bin, MKS folder  and then turned on the power.  The machine did not seem to do anything different than it's normal booting process.  Unfortunately now when I press buttons on the display it seems that even though the command is recognized it is not accepted and the machine does nothing.  I just want to put it back to original firmware and cannot do that.

reynol3232, 2022-03-15:
Much appreciated.  Yes, I asked Artillery for the stock files.  Not sure how I'm going to load them back on though when I receive the files.  The bootload is not seeming to read the microsd but I could be wrong.  I'd really like to try a stock firmware.bin file from this latest version of Genius Pro but I can't seem to find one online.  Is there a way to extract the firmware.bin file from another machine?  

Can I buy the older Touchscreen and replace the newer version so that I can program it?

reynol3232, 2022-03-15:
Starting over:  The Genius Pro "stock" firmware platformio.ini will not compile in VSC code. 
*** [.pio\build\ARTILLERY_RUBY\firmware.elf] Error 1
Made sure there were no "- or spaces" in the file directory. Any idea what I'm doing wrong?  Anyone want to send me a stock compiled firmware?  I think it's time to switch to Prusa.

stormz85, 2022-03-16:
Hi ! 
Any news for GD32* TFT ? on the SX2 ? 
Thx

reynol3232, 2022-03-17:
I was able to get my "New Version GD32F TFT" Genius Pro running again.  It was quite involved and I'm sorry I tried to upgrade the firmware.  Staunch warning to anyone trying to upgrade the firmware, wait a few months for the GD32F to be supported.

PrEzi3, 2022-03-17:
@reynol3232  it would be great if you could post the stock files somewhere --- just in case there are more ppl doing a flashing by mistake. Thanks !

rawlus, 2022-03-20:
Artillery has a github repo with updated Marlin, Genius Pro Firmware and TFT Firmware for a genius pro here:

https://github.com/artillery3d/genius-pro-firmware
https://github.com/artillery3d/genius-pro-tft-firmware

PrEzi3, 2022-03-20:
> Artillery has a github repo with updated Marlin, Genius Pro Firmware and TFT Firmware for a genius pro here:
> 
> https://github.com/artillery3d/genius-pro-firmware https://github.com/artillery3d/genius-pro-tft-firmware

....which is a precompiled binary and not the source code....  but oh well.... at least something...

reynol3232, 2022-03-20:
I've had so much trouble I think I'm going to wait for the community debug this before I try it. My machine must make parts right now.

ETE-Design, 2022-04-22:
@kisslorand Mabye I can get some info, but not whole Source... So what Info is needed to get this working?

freakydude, 2022-04-22:
@digant73 Would try some beta firmware for you. But please give me some short description how to flash and how to flash back in my case. Let me know. Thx

ark-, 2022-05-06:
https://github.com/artillery3d/sidewinder-x2-tft-firmware/commit/10dcf32da803dbb95262f46133c55ad57d2a1440

Artillery just added a commit with a new binary to their X2 TFT firmware repo. The commit message reads: "added support for GD32F305 TFT screen"

Wondered if this would be of use to anyone here.

ETE-Design, 2022-05-06:
@ark- Think people here are most interested in the Source Code, caise then it can mabye be integrated with BBT TFT Firmware :-)

ark-, 2022-05-13:
Yeah ok, just thought it was an interesting note that they added something. 

Regarding my question about determining if my X2 is GD32F305 or not, is there a way to do it without opening the unit up?

HeterodyneN7, 2022-05-13:
> Yeah ok, just thought it was an interesting note that they added something.
> 
> Regarding my question about determining if my X2 is GD32F305 or not, is there a way to do it without opening the unit up?

Yeah, get a cheap usb endoscope(they are like 5 eur) from aliexpress, I already had one on hand so it was easy to confirm which chip I had.

reynol3232, 2022-05-13:
Just open the case.  If you have trouble with the machine Artillery will ask you to take it apart and troubleshoot with them via chat.  Also, the warranty is a thing of fiction when you buy a Chinese printer.  You essentially own it once they charged your credit card.  They will send parts and/or a new machine.  Take a picture of the unbroken seal first if it make you feel better.  I've bee through it a few times.

freakydude, 2022-07-18:
Any news about flashing the display on Artillery Sidewinder X2 with GD32F305? I still don't know any option.

Thx

SudoMolto, 2022-07-25:
Some news about custom firmware on Artillery Sidewinder X2 with GD32F305?

reynol3232, 2022-09-24:
Unfortunately there has been no action on this issue. I love my Artillery machines but I very much dislike the GUI. I bought a Mingda Magician X and I love their GUI. The Minda does not print nearly as nice but it makes me wonder why Artillery doesn't upgrade their GUI. 

I'll patiently await support for this my TFT.

freakydude, 2022-09-24:
A guy named 3deann, posted me on my thingiverse project [(here)](https://www.thingiverse.com/thing:5359311/comments) the following:

> Hello!
GD32F305 free programm for Windows
Chip manufacturer: gigadevice.com/support/tech-support/
Download: www.gd32mcu.com/en/download/7
GD32 All-In-One Programmer

> Every Sidewinder X2 and Genius Pro user, programmer could test it and write progress in a thread.
I think probably the STM32 files can be loaded and saved also in the GD32 Programm?
Or the contents of the STM32 files could build up to GD32 step by step.

> The printer manufacturer people having all the actual firmware programms on their pc`s since many month.

> So if we work together with all programmers, an actual Tft-Firmware should be possible on many more creative, different ways.
Great work! Feel free to comment.

I did not have time yet to look into this, but the description sounds promising.

Best regards, freakyDude

Pontifex42, 2022-10-07:
I'm a bit confused by all the links given above. Is there a solution for the GD32* TFT issue or not? Does any successful upgrade on a machine with this TFT exist or not?
I once tried that wrong stuff and it took me a whole day to get the Sidewinder working again, will not go through it once more.
And if there is no possibility to use a decent firmware on the X2 after a year of waiting, it is time to look for a serious printer brand. The X2 made that much trouble to me ... ok, it is a 3D printer, it is its job to make trouble, but .... there must be a printer beside the Prusa which works more or less with a up-to-date marlin.

reynol3232, 2022-10-07:
I agree. There is no solution that I'm aware of and have gone through the same pain. I've been searching for the previous version screen STM...107 and they can be had for about $40. I've decided that it's just not worth it to me. I have several other printers including Prusa MK3S+ but the Artillery still cranks out the best prints. I use the Prusa for PETG and other materials but the wall finish is superior from the Artillery. Other options are to install Klipper but that's expensive as well. If I'm going to upgrade I'll likely put the money into linear rails and a bond tech extruder for the Prusa.  My next printer will be a Prusa Clone with upgrades. More money than the Artillery Genius but I hate Artillries firmware that much and it frustrates me that I can't upgrade it due to the unsupported TFT. In particular I hate that is not a single button to preheat the bed and nozzle. Good luck

freakydude, 2022-10-07:
No solution without someone how is willing and able to programm this touch screen controller. Thats why the informations are posted here. 

I personally don't use it and removed the tft from printer. Just using klipper with mobile devices and/or pc/laptop. Works great. 

If you like to use the printer with a touch screen and marlin, then buy a supported one an connect it to the mainboard. That works too.

trick76cj5, 2022-11-25:
I see the new BTB TFT's are shipping with the GD32F205 chips . . . would the firmware for these boards now be applicable to those of us w/an X2 and the GD32F305 boards?   @digant73 

https://biqu.equipment/collections/lcd/products/btt-tft43-v3-0-tft50-v3-0-tft70-v3-0-display-touch-screen-two-working-modes

Sorry, i'm a ME by trade, i'm trying to learn coding . . . so this may be a totally non-value-added mention.

reynol3232, 2022-12-08:
Does anyone know if I can simply buy the old ST32* TFT screen and install in in my newer Genius Pro?  I'm thinking this would allow me to load dignant73 firmware.  I hate the stock firmware that come with the Genius Pro.  Primarily I hate that there is no Preheat button.  You have to select the heat of the nozzle and select the heat of the bed by touching the +/- buttons dozens of times.

ark-, 2022-12-14:
I have read online people doing this with the X2. So maybe? My understanding is it just sends serial G codes, so chances look good.

reynol3232, 2022-12-16:
Ark- thanks for the reply. Do you happen to recall where you read others doing what I proposed? I'd like to possibly inquire further on that posting. 

Thanks

ciotto, 2023-01-09:
I'm a (un)fortunated owner of an _Artillery Genius Pro_ with **GD32F305VC** TFT screen.

I tried to do some tests and the reason why the [original firmware ](https://github.com/artillery3d/genius-pro-tft-firmware) update starts without issue is the name of the file.

In fact, if you try to update without the `mkstft28evo.bin` file, introduced by [this commit](https://github.com/artillery3d/genius-pro-tft-firmware/commit/3abde719d25230af3ef90d874a1365de1a931962), nothing happens, precisely like the update with this custom firmware.

This means that the update start only using the `mkstft28evo.bin` filename, but trying without more information can brick my LCD, so it's only a suppose.

The previous version of the board used an **STM32F401RC** MCU.

[image]

**GD32F305VC** and **STM32F401RC** are very similar MCUs, in fact both are Arm® Cortex®-M4 32-bit MCU, but somethings is different:

| Check | Features | **[GD32F305VC](https://datasheetspdf.com/pdf-file/1481673/GigaDevice/GD32F305VCT6/1)** | **[STM32F401RC](https://www.st.com/resource/en/datasheet/stm32f401rc.pdf)** |
|---|---|---|---|
| [x] | MCU type | M4 | M4 |
| [ ] | Flash memory | 1024 KB  | 256 KB |
| [ ] | SRAM | 96 KB | 64 KB |
| [ ] | Frequency | 120 MHz  | 84 MHz |

Also, seems that the pinout is more or less the same:

[image]

So I can suppose that the board is more or less the same.

The good news is that the **GD32F305VC** is more powerful, the bad news is that probably requires a different compiled binary.

I'm not an expert, but probably I can try to add a new configuration to Platformio for **MKS TFT28 Evo**.

SudoMolto, 2023-01-09:
> I'm a (un)fortunated owner of an _Artillery Genius Pro_ with **GD32F305VC** TFT screen.
> 
> I tried to do some tests and the reason why the [original firmware ](https://github.com/artillery3d/genius-pro-tft-firmware) update starts without issue is the name of the file.
> 
> In fact, if you try to update without the `mkstft28evo.bin` file, introduced by [this commit](https://github.com/artillery3d/genius-pro-tft-firmware/commit/3abde719d25230af3ef90d874a1365de1a931962), nothing happens, precisely like the update with this custom firmware.
> 
> This means that the update start only using the `mkstft28evo.bin` filename, but trying without more information can brick my LCD, so it's only a suppose.
> 
> The previous version of the board used an **STM32F401RC** MCU.
> 
> [image]
> 
> **STM32F401RC** and **GD32F305VC** are very similar MCUs, in fact both are Arm® Cortex®-M4 32-bit MCU, but somethings is different:
> Check 	Features 	**[STM32F401RC](https://www.st.com/resource/en/datasheet/stm32f401rc.pdf)** 	**[GD32F305VC](https://datasheetspdf.com/pdf-file/1481673/GigaDevice/GD32F305VCT6/1)**
> [x] 	MCU type 	M4 	M4
> [ ] 	Frequency 	84 MHz 	120 MHz
> [ ] 	Flash memory 	256 KB 	1024 KB
> [ ] 	SRAM 	64 KB 	96 KB
> 
> Also, seems that the pinout is more or less the same:
> 
> [image]
> 
> So I can suppose that the board is more or less the same.
> 
> The good news is that the **GD32F305VC** is more powerful, the bad news is that probably requires a different compiled binary.
> 
> I'm not an expert, but probably I can try to add a new configuration to Platformio for **MKS TFT28 Evo**.
-Se ci riesci sarebbe veramente fantastico! 
-If you can that would be great.

Fourmi, 2023-01-20:
> I'm a (un)fortunated owner of an _Artillery Genius Pro_ with **GD32F305VC** TFT screen.
>
> I tried to do some tests and the reason why the [original firmware [...]

The problem is not the Ruby card chip but the TFT controller chip.
On my side I integrated the GD32F305 chip libraries in the Bigtreetech firmware, I can compile the firmware but I can't upload it on the Sidewinder TFT.
I can share my work, if it can help 
Thanks
[image]

ciotto, 2023-01-23:
Great work @Fourmi, if I have some time I'll try.

My guess is that you can upload the firmware of your **Sidewinder TFT** by using the filename `mkstft28evo.bin`, that is the one that my Genius Pro read from the original firmware update.

If you are confident and you can try, please share with us the result!

Fourmi, 2023-01-24:
Yes, we can try by renaming the file and see if it works.
It's frustrating not to know why it's blocked at this stage.
I spent a lot of time on this, I used the existing structure with the existing GD32F205 chip in this firmware.

Dave811, 2023-01-24:
I tired both of your repos (first flashed with original to test that my SD-Card is working, and after that I used your repos to compile it), but both dont work on my end :/

ciotto, 2023-01-24:
@Dave811 you use `mkstft28evo.bin` as filename?

Dave811, 2023-01-24:
@ciotto yes I used that name, and also in all uppercase, just in case. But both dont work.

Fourmi, 2023-01-24:
@Dave811 , yes, compile is ok ,but upload don't load with SDcard
it is this point that should be seen and which blocks me to advance a little more

ciotto, 2023-01-25:
Finally **I was able to update the TFT firmware** with the one compiled by @Fourmi (I have a video), but in the end, nothing worked.

The trick is to use the [original firmware](https://github.com/artillery3d/genius-pro-tft-firmware) as a base, replacing the `mkstft28evo.bin` and adding other BIGTREETECH firmware files, so we have:

- [mks_font/](https://github.com/artillery3d/genius-pro-tft-firmware/tree/main/mks_font)
- [mks_pic/](https://github.com/artillery3d/genius-pro-tft-firmware/tree/main/mks_pic)
- [MksWifi.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/MksWifi.bin)
- [mks_config.txt](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mks_config.txt)
- [mkstft28.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mkstft28.bin)
- [mkstft28evo.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mkstft28evo.bin) (replaced by [GD_TFT28_X2.27.x.bin](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/GD_TFT28_X2.27.x.bin))
- [TFT28](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme/TFT28)
- [config.ini](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini)

The procedure requested me to do e touchscreen calibration, but I can't do anything. I suppose that the pin mapping is wrong.

[image]

After restarting, the screen shows the BIGTREETECH logo but nothing else.

Now I re-flashed hte original firmware and all works fine.

I think that it's a great result, now we can fix the firmware.

I'll update you ASAP with more details and some images.

ziirish, 2023-01-25:
@Fourmi can you provide a bit of details on how you compiled the firmware? I tried to compile it based on your repo but I'm encountering the following error:

```
$ pio run
Processing MKS_TFT28_X2 (board: GD32F305VC_0X3000; platform: https://github.com/CommunityGD32Cores/platform-gd32.git; framework: spl)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'GD32F305VC_0X3000'
```

Thanks

ciotto, 2023-01-25:
I want to try to do some changes but after the first installation, the process never reaches the end.

The first time, the process was this:

- original firmware update
  - updating
[image]

  - SPI Flash Erasing
[image]

  - Erasing success
  - Load PIC
    - copy `_bmp_*.bin` files
[image]

  - Load font
    - copy 2 `0:/mks_font/*.bin` files
  - Complete
[image]

- BTT firmware update
  - Updating fonts
    - `SD:TFT28/font/*.fon` files
  - Updating Icons
    - Copy S`D:TFT28/bmp/*.bmp` files (with preview)
[image]

  - Updating Configuration...
[image]

  - Touch Screen Calibration

Now the update stop after **Complete** screen, I don't know why.

Does someone else have the same problem?

I'll do more test.

nomc2, 2023-01-25:
> Finally **I was able to update the TFT firmware** with the one compiled by @Fourmi (I have a video), but in the end, nothing worked.
> 
> The trick is to use the [original firmware](https://github.com/artillery3d/genius-pro-tft-firmware) as a base, replacing the `mkstft28evo.bin` and adding other BIGTREETECH firmware files, so we have:
> 
> * [mks_font/](https://github.com/artillery3d/genius-pro-tft-firmware/tree/main/mks_font)
> * [mks_pic/](https://github.com/artillery3d/genius-pro-tft-firmware/tree/main/mks_pic)
> * [MksWifi.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/MksWifi.bin)
> * [mks_config.txt](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mks_config.txt)
> * [mkstft28.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mkstft28.bin)
> * [mkstft28evo.bin](https://github.com/artillery3d/genius-pro-tft-firmware/blob/main/mkstft28evo.bin) (replaced by [GD_TFT28_X2.27.x.bin](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/GD_TFT28_X2.27.x.bin))
> * [TFT28](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme/TFT28)
> * [config.ini](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini)
> 
> The procedure requested me to do e touchscreen calibration, but I can't do anything. I suppose that the pin mapping is wrong.
> 
> [image]
> 
> After restarting, the screen shows the BIGTREETECH logo but nothing else.
> 
> Now I re-flashed the original firmware and all works fine.
> 
> I think that it's a great result, now we can fix the firmware.
> 
> I'll update you ASAP with more details and some images.

Did not work for me on my X2, I get stuck at a black screen that says complete and the TFT never reboots.
Are you sure you just replaced the mkstft28evo.bin file and not something else as well?

ciotto, 2023-01-25:
I'm pretty sure, but now I have the same problem, so probably something else is different... 😩

ciotto, 2023-01-27:
One more update.

I tryed the procedure using the MKS_TFT28_V4.0.27.x.bin file and doesn't stop after the "Completed" screen, but this is what I see.

[image]

As usual, after I can flash the original firmware.

ciotto, 2023-01-28:
I'm trying a new approach. Starting from the STM32 version of the firmware, I have a "working" firmware that shows the BTT logo:

[image]

Starting from this, I want to try some debugging.

If you want to try, [here](https://github.com/ciotto/BIGTREETECH-TouchScreenFirmware/tree/MKS-GD-TFT28-release/mkstft28evo) is available the "working" firmware with all the file and folder needed for uploading.

FabulosoDev, 2023-01-29:
> I have a "working" firmware that shows the BTT logo
Promising progress you made 👍. Could you please describe what 'working' means in that context?
Is it the regular STM firmware you were able to compile for the GD32 MCU?
Thanks for your reply.

ciotto, 2023-01-30:
As I said, "working" means that it "shows the BTT logo", literally 🙂.

This version was essentially built using STM32 libraries, after flashing show the logo screen with the `Card inserted` and `Card removed!` strings (as you can see in my previous post).

After working on it for a day, now I reached the same result with a version built using the GB32 libraries.

This version has some screen glitches and is slower, I don't know why (yet).

In any case, for both versions the problem that stop the execution on boot screen is this loop on [ModeSwitching.c](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/master/TFT/src/User/API/ModeSwitching.c#L38):

```
while (OS_GetTimeMs() - startUpTime < BTT_BOOTSCREEN_TIME)  // display logo BTT_BOOTSCREEN_TIME ms
{
  loopProcess();
}
```

The loop never ends due to the fact that `OS_GetTimeMs()` is always `0`. Probably it's related to the timer interrupt call.

Ricky1966, 2023-02-15:
@ciotto 
I need to write to you but the message was returned to your site's email address.

ciotto, 2023-02-15:
@Ricky1966 I fixed the issue with my email and now you can write me.

Ricky1966, 2023-02-16:
@ciotto 
If you think problem nested in the interrupt 
OS_GetTimeMs(), have you try to replace this value with a fixed as 1000 (1 second)?

MirakelPrints, 2023-02-16:
Sadly this is not a fixed time. Its a counter incremented on interrupt.
See [os_timer.c#L76](https://github.com/Fourmi/BIGTREETECH-TouchScreenFirmware-master/blob/cd2c523a610f9d48cddb84e85eb94757a5fbf565/TFT/src/User/os_timer.c#L76)

[553 more replies]

PR #2886 (2024-02-19): Add support for MKS GB boards.
### Requirements

To test this fix you need a MKS board with GigaDevice MCU (used on Artillery Genius Pro and other Artillery 3D printers). 

### Description

As you can see [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2391), the new versions of the MKS boards use GigaDevice MCUs.

We know that there are at least 2 versions of the board that use two different versions of the bootloader:

 - [x] TFT Controller v1.2. with bootloader V3.0.4 (tested)
 
[image]

 - [x] TFT Controller v1.4 with [bootloader V3.0.5](https://github.com/ciot [... cut]
closed 2024-02-19 by bigtreetech