bigtreetech/BIGTREETECH-TouchScreenFirmware #2398: Embedded Gcode thumbnails

trizmark, 2022-02-15:
I am using PrusaSlicer to generate my Gcode files and unable to get thumbnails to display on the BTT TFT35.

I have modified the PrusaSlicer configuration and enabled the G-code thumbnails and set it to 95x95.
I have changed Configuration.h and set THUMBNAIL_PARSER to 2. After compilation I can see that the RAM usage goes from 20028 bytes to 63940 bytes (which is in line with what's required for the Base64 decoding).

When I browse the files I can see a split second wait (I assume trying to process the embedded thumbnail), but no thumbnail is displayed.

Do I need to do anything else to get thumbnails working?

sarvenn, 2022-02-15:
Definately there is one way else and it's working perfect!
https://github.com/effgarces/Biqu-Thumbnail-Generator

I use the thumbnail parsel 0 (default-classic). I use this post processing script (it is now compatible with PS2.4 also) , it's flawless.

Download this ( https://github.com/effgarces/Biqu-Thumbnail-Generator/releases/download/0.03/biqu_convert_p24.zip ) and copy the standalone script to your prusaslicer.exe folder. Then add the script name to PS just like this:
[image]

sarvenn, 2022-02-17:
News,
Hopefully it would be avaible in Superslicer without the need of script usage.
Merill is working on it.
https://github.com/supermerill/SuperSlicer/issues/887

trizmark, 2022-02-18:
Finally I had some time to look into what's going on with the firmware. After a bit of debugging, I realised that the only reason why the base64PNG thumbnail is not displayed is because the code is looking for it in the wrong place.

Let me elaborate:
First I set the THUMBNAIL_PARSER to 2 to ensure that we're looking for base64PNG thumbnails
After generating a gcode file with PrusaSlicer (with thumbnail size set to 95x95) I expected the firmware to do the following:
- look for base64PNG,
- look for RGB565 bitmap in comments
- look for RGB565 bitmap at fixed location

Looking at the debug logs, the order is exactly the opposite.
The main issue is that looking for RGB565 bitmap in comments move the file pointer (as the function `bool modelFileFind(FIL *fp, char *find)` reads the file) to `MAX_THUMBNAIL_SEARCH_BLOCKS`. Once its established that RGB565 thumbnail can be found, the code goes on to look for base64PNG, but from the current position! Which, of course, will fail as the comment is at the beginning of the file.

The fix is super-simple. We simply need to seek back to the beginning of the file before we start looking for the base64PNG comment signature. I'll prep a PR for this.

trizmark, 2022-02-18:
Closing as the issue is clear and a fix has been found.

closed 2022-02-18 by trizmark