bigtreetech/BIGTREETECH-TouchScreenFirmware #2401: [BUG] MKS TFT35 V1.0 support doesn't work

Sanerok, 2022-02-16:
So, after adding support of MKS TFT 35 I compiled sources and flashed my screen.
Also, I renamed .bin to MKSTFT35.bin and located it to SD card as well with MKS folder and config .ini file.  My screen after flashing looks like on the attached files:

https://user-images.githubusercontent.com/38011897/153462306-24d43428-7a1e-444e-aeea-c0ee3d3197cb.mov
https://user-images.githubusercontent.com/38011897/153462323-f89a68ba-7211-49ad-9893-631922e7fcba.jpeg

digant73, 2022-02-16:
@did you provide also the MKS folder with icons and fonts?

Sanerok, 2022-02-16:
**Case 1:**
Renamed MKS folder to TFT35
Use encrypted bin and default config 
**OUTPUT:** Screen freezes on "Updating screen":
[image]

**Case 2:**
Renamed MKS folder to TFT35
Use **un**encrypted bin and default config 
**OUTPUT:** After uploading, screen looks like this:
https://user-images.githubusercontent.com/38011897/154260229-166b7327-7a93-4a55-84d0-e2b2f674442b.mov

The tree output looks like this for both cases:
.
├── config.ini
├── MKSTFT35.bin
└── TFT35
    ├── bmp
    │   ├── 001_mm.bmp
    │   ├── 01_mm.bmp
    │   ├── 100_mm.bmp
    │   ├── 10_degree.bmp
    │   ├── 10_mm.bmp
    │   ├── 1_degree.bmp
    │   ├── 1_mm.bmp
    │   ├── 5_degree.bmp
    │   ├── apply.bmp
    │   ├── babystep.bmp
    │   ├── back.bmp
    │   ├── baud_rate.bmp
    │   ├── bed.bmp
    │   ├── bltouch.bmp
    │   ├── bltouch_deploy.bmp
    │   ├── bltouch_repeat.bmp
    │   ├── bltouch_reset.bmp
    │   ├── bltouch_stow.bmp
    │   ├── bltouch_test.bmp
    │   ├── brightness.bmp
    │   ├── case_light.bmp
    │   ├── chamber.bmp
    │   ├── connection_settings.bmp
    │   ├── cooldown.bmp
    │   ├── custom.bmp
    │   ├── dec.bmp
    │   ├── delta_calibrate.bmp
    │   ├── disable_steppers.bmp
    │   ├── disconnect.bmp
    │   ├── e_100_mm.bmp
    │   ├── e_10_mm.bmp
    │   ├── e_10_percent.bmp
    │   ├── e_1_mm.bmp
    │   ├── e_1_percent.bmp
    │   ├── e_200_mm.bmp
    │   ├── e_5_mm.bmp
    │   ├── e_5_percent.bmp
    │   ├── eeprom_reset.bmp
    │   ├── eeprom_restore.bmp
    │   ├── eeprom_save.bmp
    │   ├── extrude.bmp
    │   ├── fan.bmp
    │   ├── fan_full_speed.bmp
    │   ├── fan_half_speed.bmp
    │   ├── fast_speed.bmp
    │   ├── feature_settings.bmp
    │   ├── file.bmp
    │   ├── folder.bmp
    │   ├── gcode.bmp
    │   ├── global_bed.bmp
    │   ├── global_chamber.bmp
    │   ├── global_nozzle.bmp
    │   ├── heat.bmp
    │   ├── heat_fan.bmp
    │   ├── home.bmp
    │   ├── home_move.bmp
    │   ├── inc.bmp
    │   ├── infobox.bmp
    │   ├── language.bmp
    │   ├── level_corner.bmp
    │   ├── level_edge_distance.bmp
    │   ├── leveling.bmp
    │   ├── leveling_off.bmp
    │   ├── leveling_on.bmp
    │   ├── load.bmp
    │   ├── logo.bmp
    │   ├── logo_portrait.bmp
    │   ├── machine_settings.bmp
    │   ├── mainmenu.bmp
    │   ├── manual_level.bmp
    │   ├── marlin_mode.bmp
    │   ├── mesh_editor.bmp
    │   ├── mesh_valid.bmp
    │   ├── more.bmp
    │   ├── move.bmp
    │   ├── normal_speed.bmp
    │   ├── nozzle.bmp
    │   ├── nozzle_down.bmp
    │   ├── nozzle_up.bmp
    │   ├── onboard_sd.bmp
    │   ├── ontft_sd.bmp
    │   ├── page_down.bmp
    │   ├── page_up.bmp
    │   ├── parameter.bmp
    │   ├── pause.bmp
    │   ├── percentage.bmp
    │   ├── pid.bmp
    │   ├── point_1.bmp
    │   ├── point_2.bmp
    │   ├── point_3.bmp
    │   ├── point_4.bmp
    │   ├── point_5.bmp
    │   ├── preheat.bmp
    │   ├── preheat_both.bmp
    │   ├── print.bmp
    │   ├── printing_bed.bmp
    │   ├── printing_chamber.bmp
    │   ├── printing_fan.bmp
    │   ├── printing_flow.bmp
    │   ├── printing_nozzle.bmp
    │   ├── printing_speed.bmp
    │   ├── printing_timer.bmp
    │   ├── printing_zlayer.bmp
    │   ├── probe_offset.bmp
    │   ├── readme.md
    │   ├── reset_value.bmp
    │   ├── resume.bmp
    │   ├── rgb_blue.bmp
    │   ├── rgb_green.bmp
    │   ├── rgb_off.bmp
    │   ├── rgb_red.bmp
    │   ├── rgb_settings.bmp
    │   ├── rgb_white.bmp
    │   ├── rotate_ui.bmp
    │   ├── screen_info.bmp
    │   ├── screen_settings.bmp
    │   ├── settings.bmp
    │   ├── shut_down.bmp
    │   ├── slow_speed.bmp
    │   ├── sound.bmp
    │   ├── status_bed.bmp
    │   ├── status_chamber.bmp
    │   ├── status_fan.bmp
    │   ├── status_flow.bmp
    │   ├── status_nozzle.bmp
    │   ├── status_speed.bmp
    │   ├── stop.bmp
    │   ├── touch_mode.bmp
    │   ├── touchscreen_adjust.bmp
    │   ├── tune_extruder.bmp
    │   ├── tuning.bmp
    │   ├── u_disk.bmp
    │   ├── unload.bmp
    │   ├── x_dec.bmp
    │   ├── x_home.bmp
    │   ├── x_inc.bmp
    │   ├── y_dec.bmp
    │   ├── y_home.bmp
    │   ├── y_inc.bmp
    │   ├── z_align.bmp
    │   ├── z_dec.bmp
    │   ├── z_fade.bmp
    │   ├── z_home.bmp
    │   └── z_inc.bmp
    └── font
        ├── 8x16_byte_ascii.fon
        ├── byte_ascii.fon
        ├── large_byte_ascii.fon
        ├── readme.md
        └── word_unicode.fon

digant73, 2022-02-16:
as far as I see it seems wrong bmp and fonts are used. The "TFT35" folder must contain the "bmp" and "font" folder. From your output, it seems bmp and font folders are at root level

Sanerok, 2022-02-16:
Just updated with proper pics and fonts and now everything work fine. "Copy to SD Card - MKS" folder disoriented me, so I used wrong resources. Thanks for helping.

digant73, 2022-02-16:
ok. In any case in my current PR, I will change the "TFT35" folder to "MKS" (the same used also for TFT28) so the README.md file will not need to be changed. Remember that in the future.
Close this ticket if the issue is solved

Sanerok, 2022-02-16:
I compiled this one, but resources maybe was from another one.

closed 2022-02-16 by Sanerok