bigtreetech/BIGTREETECH-TouchScreenFirmware #2403: [BUG] Hard to update everything after PR #2380

kisslorand, 2022-02-16:
### Description

After PR #2380 the full update of the firmware is very cumbersome. Files get deleted before they are read, backup files (.CUR) are also deleted.

### Steps to reproduce

1. Copy bitmaps, fonts, firmware file, language file, config file to SD Card
2. Reboot TFT

**Expected behavior**
I expect everything to be updated.

**Actual behavior**
 - If icons, fonts, firmware file, language file, config file are new, only fonts, firmware and language file is updated. "config.ini" and, if it was present, "config.ini.CUR" is deleted from SD Card
 - If firmware file and config file is only present, upon reboot only FW updates, "config.ini" gets deleted
 - if only "config.ini" and "language_xx.ini" is present, both get updated but "config.ini.CUR" is not present (not created or deleted if it was there before)
 - icons never get updated
 - config can be updated only if there's no firmware file
 - inf config is updated the firmware backup (.CUR) gets deleted
 - if only firmware is updated config backup (.CUR) gets deleted
 - if only language is updated config backup (.CUR) gets deleted

There can be other quirks but these were enough for me to call it a bug.

### Hardware Variant
BTT TFT35 v3

### TFT Firmware Version & Main Board Firmware details
Latest TFT firmware as of today, SKR v2 (F429) with latest Marlin bugfix as of today.

digant73, 2022-02-17:
I can confirm that in case config.ini.CUR is already present and a language file is updated then config.ini.CUR is deleted by function getLangFromFile (probably during the renaming of the language file).
I don't know if the issue was already present in the past

kisslorand, 2022-02-17:
Today I was also able to update all the elements at once.. The thing I did different today was to not have the USB stick connected at all to the TFT.

PR #2402 (2022-02-18): Some code reduction + cleanup
**IMPROVEMENTS:**
* Some code reduction + cleanup after the last merged PRs
* Used TFT* (e.g. TFT28) folder naming convention also for MKS TFTs: There is no more need to rename to MKS folder. README.md file updated accordingly

**BUGFIXES:**
* Fixed missing check on source filename before trying to rename to .CUR during boot: Small bug originated in #2380.

**NOTE:** Reporting #2295 as fixed just to close it (it was fixed by other merged PRs).

fixes #2403
fixes #2295

**PR STATE:** Ready for merge