bigtreetech/BIGTREETECH-TouchScreenFirmware #2405: [BUG] Display the number of layers

radek8, 2022-02-17:
### Description
The print status screen shows the layer height and the layer sequence number. With the latest updates, it has shattered.
@kisslorand, can you please see where it went?
I use Cura where the first layer is labeled 0

**Expected behavior**

It always appeared to me when I was printing
height 0.2mm, - layer 1/10
height 0.4mm - layer 2/10
...
height 2.00mm - layer 10/10

**Actual behavior**

It has changed since the last update.
height 0.2mm - layer ---
height 0.4mm - layer 1/10
...
height 2.00mm - layer 9/10

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

TFT35 E3 V3.0

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2096

kisslorand, 2022-02-17:
Check PR #2404 or replace in "comment.c"
`else if (temp_char[0] - '0' && temp_char[0] <= '9') // check if a number is found`
with
`else if (temp_char[0] >= '0' && temp_char[0] <= '9') // check if a number is found`

radek8, 2022-02-17:
Thank you.
I'm printing, I'll test when printing is over.

Guilouz, 2022-02-17:
Also issue when number of layer exceed 1000 layers, TFT only show actual layer and not as example 864 / 1248.
Under 1000 layers no issue.

Guilouz, 2022-02-17:
> That's by design. There's no room for 8 digits and a dash ("/").

On TFT70, I think there is enough space

kisslorand, 2022-02-18:
> I'm printing, I'll test when printing is over.

Long print... :)

kisslorand, 2022-02-18:
> > That's by design. There's no room for 8 digits and a dash ("/").
> 
> On TFT70, I think there is enough space

Replace in "PrintingMenu.c":
`    if (layerCount > 0 && layerCount < 1000)`

with

```
if (layerCount > 0
#ifndef TFT70_V3_0
  && layerCount < 1000
#endif
)
```

Please test it and report back

radek8, 2022-02-18:
> > I'm printing, I'll test when printing is over.
> 
> Long print... :)

I apologize for the delay. The press ended late at night and I had to go to work in the morning.
I applied your suggested edit and it solved the layer numbering problem.
Thank you

rondlh, 2022-02-19:
How about making it alternate between the current layer number and layer count like is done with the flow and speed icon?

This line of code seems very odd to me:
printingmenu.c, line 142
`sprintf(layer_number_txt, "%s%u%s", "  ", layerNumber, "  ");`

Isn't this the same as:

`sprintf(layer_number_txt, "  %u  ", layerNumber);`

kisslorand, 2022-02-19:
Theoretically is the same, practically not. The version suggested will have different result than expected.

> How about making it alternate between the current layer number and layer count like is done with the flow and speed icon?

It was a debate how to handle this already and there were many suggestions. The current behaviour is the result of that debate.

rondlh, 2022-02-19:
Thanks for you quick reply!
About the sprintf question... could you enlighten me by showing me the difference in the resulting string (layer_number_txt) would be?
I tried it with this code
```
  char layer_number_txt1[14];
  char layer_number_txt2[14];
  sprintf(layer_number_txt1, "%s%u%s", " ", 999, " ");
  sprintf(layer_number_txt2, " %u ", 999);
  int result = strcmp(layer_number_txt1, layer_number_txt2);
  printf("strcmp(layer_number_txt1, layer_number_txt2) = %d\n", result);    
```
strcmp says the resulting strings are equal.

I'm currently testing this code (BTT TFT35 V3.0):
```
else
    {
      static bool alternate = false;
      //sprintf(layer_number_txt, "%s%u%s", "  ", layerNumber, "  ");
	  if (alternate) 
	    sprintf(layer_number_txt, "  %.3u/ ", layerNumber);
	  else
	    sprintf(layer_number_txt, "  /%.3u ", layerCount);
      alternate = !alternate;	
    }
```
Seems to work fine up till now... waiting for layer 1000 to come around.

kisslorand, 2022-02-19:
The form you suggested gave me some headache, it didn't want to delete characters at the end if previous number had more digits than actual one. That's the reason why " " characters are there, to delete remaining garbage in front and at the end of the new number/text to be displayed. I cannot remember on what TFT happened as I have more with different MCUs.

radek8, 2022-02-19:
Yes, if you started a new print, fragments of the previous value remained there.

rondlh, 2022-02-19:
Please note that I leave the space in place, I just don't hand the space to the sprintf statement as parameters, I just enter them into the format string directly... (note the space before and after the %u)
`sprintf(layer_number_txt, " %u ", layerNumber);`

rondlh, 2022-02-19:
The code seems to work fine for me. It alternates every layer.
```
else
    {
      static bool alternate = false;
      //sprintf(layer_number_txt, "%s%u%s", "  ", layerNumber, "  ");
	  if (alternate) 
	    sprintf(layer_number_txt, "  %.3u/ ", layerNumber);
	  else
	    sprintf(layer_number_txt, "  /%.3u ", layerCount);
      alternate = !alternate;	
    }
```

[image]
[image]

kisslorand, 2022-02-19:
> The code seems to work fine for me. It alternates every layer.
>
> ```
> else
>     {
>       static bool alternate = false;
>       //sprintf(layer_number_txt, "%s%u%s", "  ", layerNumber, "  ");
> 	  if [...]

As I said, I've been there. " %u " was my first obvious choice.

rondlh, 2022-02-19:
Of course, I fully agree, thanks!

closed 2022-03-21 by bigtreetech