bigtreetech/BIGTREETECH-TouchScreenFirmware #2411: Problem Inverted Letters Help:((

sokolek155, 2022-02-19:
I have problem my BTT TFT 35 v3.0 i have inverted letters.

oldman4U, 2022-02-19:
ReadMe from front page - Troubleshooting section.

Please help the community and close the ticket once you do not need it anymore. Thank you

sokolek155, 2022-02-19:
![20220219_125311](https://user-images.githubusercontent.com/84595174/154804110-08e78a45-23cf-4c0e-b4a8-6941776133c2.jpg

digant73, 2022-02-19:
put a blank file named **reset.txt** in the SD card and restart the TFT with the SD card in the TFT's slot. The screen calibration will start. At the end of the calibration process, the issue should be solved

sokolek155, 2022-02-19:
I did so. Screen calibrations as well. unfortunately it did not help :(

sokolek155, 2022-02-19:
The problem is that the letters are opposite not the screen.

digant73, 2022-02-19:
you have an horizontally mirrored screen. Try to install again the firmware, fonts, icon, config.ini. Are you sure you used the proper firmware? For BTT TFT35 V3, you must use BIGTREE_TFT35_V3.0.27.x.bin

radek8, 2022-02-19:
Are you sure with the display version?
Can you take a TFT picture without a cover on both sides?

sokolek155, 2022-02-19:
Exactly as you write, I have a mirror image screen. Software exactly as you wrote. Tried flashing a few times on with no results.

digant73, 2022-02-19:
As suggested by Radek, verify you have a TFT35 V3. It seems you have another version and you are installing the wrong firmware

radek8, 2022-02-19:
you send photos of the circuit board

sokolek155, 2022-02-19:
[image]
[image]
[image]

radek8, 2022-02-19:
again and do not rush

sokolek155, 2022-02-19:
[image]

sokolek155, 2022-02-19:
[image]

sokolek155, 2022-02-19:
[image]

digant73, 2022-02-19:
try to install this

[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8102552/BIGTREE_TFT35_V3.0.27.x.zip)

radek8, 2022-02-19:
It looks like some modification on the BTT side.
The STM32F207 processor was originally used
the photo shows the STM32F205 processor

sokolek155, 2022-02-19:
Working thanks you :):) Could you please write me in which line he changed the name of the processor?

radek8, 2022-02-19:
LCD is also different

[image]

[image]

sokolek155, 2022-02-19:
[image]

digant73, 2022-02-19:
> Working thanks you :):) Could you please write me in which line he changed the name of the processor?

I didn't change anything. I simply provided my compiled firmware. Maybe the precompiled firmware you used was wrongly created

digant73, 2022-02-19:
use this config.ini file

[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8102567/config.zip)

However I think you are using wrong source files (or old files)

sokolek155, 2022-02-19:
I need to upload bmp files and config files, could you send them to me so that I can add them?

digant73, 2022-02-19:
[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8102573/TFT35.zip)

sokolek155, 2022-02-19:
I recently downloaded the files from github bigtreetech I do not know why version 27 is probably the last one available.

sokolek155, 2022-02-19:
[image]

sokolek155, 2022-02-19:
Everything works fine, thank you very much for your help. A closed topic may be useful to someone. Regards and thank you.

digant73, 2022-02-19:
good you solved your issue. Please close this ticket

closed 2022-02-19 by sokolek155

radek8, 2022-02-19:
You miss download version 0.27
You have to download Master

sokolek155, 2022-02-19:
Where can i find?

radek8, 2022-02-19:
[image]

sokolek155, 2022-02-19:
I found my problem was flashing issue 27.Sorry amateur me.

radek8, 2022-02-19:
@oldman4U 

Can you write on the title page of the manual that it should be updated from the master?

digant73, 2022-02-19:
@radek8 In my current PR I made also other changes in README.md. So I can also add this info at the beginning of the document