bigtreetech/BIGTREETECH-TouchScreenFirmware #2414: [BUG] Wrong file list navigating on onboard SD

digant73, 2022-02-20:
With the introduction of support to long folder name the file list is sometime wrongly displayed.

Here a wrong list:

[image]

Here the correct list:

[image]

Attached all the content of the SD card to replicate the issue.

Tested on MKS TFT28 with 65KB RAM

kisslorand, 2022-02-20:
> Attached all the content of the SD card to replicate the issue.

Nothing attached.

Are there any specific steps to reproduce the issue?

digant73, 2022-02-20:
[sd_card_1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104661/sd_card_1.zip)
[sd_card_2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104662/sd_card_2.zip)
[sd_card_3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104663/sd_card_3.zip)

digant73, 2022-02-20:
[sd_card_4.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104664/sd_card_4.zip)
[sd_card_5.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104669/sd_card_5.zip)
[sd_card_6.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104673/sd_card_6.zip)
[sd_card_7.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104678/sd_card_7.zip)

digant73, 2022-02-20:
[sd_card_8.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104680/sd_card_8.zip)
[sd_card_9.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8104681/sd_card_9.zip)

digant73, 2022-02-20:
put all the content of the zip files in the root of the SD card

digant73, 2022-02-20:
> > Attached all the content of the SD card to replicate the issue.
> 
> Nothing attached.
> 
> Are there any specific steps to reproduce the issue?

navigate in the list menu. Sometimes I got the issue entering in the list menu. Other times, simply selecting one file and pressing "cancel" to return to the list menu etc...

kisslorand, 2022-02-20:
I am clicking here for 30 minutes and cannot reproduce it. Tested on MKS TFT28 v4 (clone) and BTT TFT35 v3.
I have a hunch... Delete a few files and see if it still happens to you.
Does it happen also from the TFT's SD Card?

digant73, 2022-02-20:
It happened only on onboard sd. I removed files (less than 40 in the root folder) and I got the the same issue. However, for the moment it's not a big problem for me. Maybe it could be the SD card not properly read by the onboard SD.

kisslorand, 2022-02-20:
I do not think it is an SD Card issue, in the first picture the first folder is a part of the  temperature autoreport:
` T:105.08 /0.00 B:27.00 /0.00 @:0 B@:0`.
It seems to be an overflow somewhere.

digant73, 2022-02-21:
Yes thanks. it fixed the issue. I also applied your changes in my vfs.c just be already aligned

closed 2022-02-21 by digant73

kisslorand, 2022-02-21:
> Yes thanks, that fixed the issue. I also applied your changes in my vfs.c just be already aligned.

Thanks for the feedback.