bigtreetech/BIGTREETECH-TouchScreenFirmware #2415: Skr2 and TFT3.5

NunoFilipePereira, 2022-02-20:
The skr2 board doesn't read my sdcard if it doesn't have a pen plugged into the board. What do I have to change in the firmware? The firmware version is marlin 2.0.9.3.

kisslorand, 2022-02-20:
This repository is for the BTT TFT firmware, not SKR v2 nor Marlin.
Please close this issue and go to the relevant repository with this problem you have.

oldman4U, 2022-02-22:
Please help the community and close the ticket. Thank you

NunoFilipePereira, 2022-02-22:
> This repository is for the BTT TFT firmware, not SKR v2 nor Marlin.
> Please close this issue and go to the relevant repository with this problem you have.

Sorry who is the manufacturer of the TFT? not Bigtreetech? Where is the error?

NunoFilipePereira, 2022-02-22:
> Please help the community and close the ticket. Thank you

Tell me what's wrong? who is the manufacturer?

oldman4U, 2022-02-22:
This is not a user group but the BTT TFT firmware repository. So if you have a feature request or you would like to report a bug related to the TFT firmware you are right. But this does not seem to be the case. So this is the wrong repository.

NunoFilipePereira, 2022-02-22:
> You are asking about Marlin, here we do not deal with Marlin.
> 
> You cannot demand help for you brand new shiny Samsung TV that doesn't play your latest pirated movie on a Samsung phone related firmware group.

No, I'm not asking for help on the marlin. I'm questioning the problem. TFT and marlin are together. Because one does not work without the other. hence the marlin issue. Tft firmware… is this where you ask? They are being very radical in a simple matter of help. But thanks. It has been the motto for the last few months of this brand

NunoFilipePereira, 2022-02-22:
> This is not a user group but the BTT TFT firmware repository. So if you have a feature request or you would like to report a bug related to the TFT firmware you are right. But this does not seem to be the case. So this is the wrong repository.

The question and this memso for what I realized and I have noticed in forums. The bug is memso in TFT in communication with Marln. There could be something new to the resolution. But ... Thanks for the help.

closed 2022-02-22 by NunoFilipePereira

radek8, 2022-02-22:
Ask here:

https://github.com/bigtreetech/SKR-2/issues