bigtreetech/BIGTREETECH-TouchScreenFirmware #2417: [BUG] Cannot start print from ESP3D(2nd serial port) for gcode on motherboard storage

rondlh, 2022-02-21:
I'm not sure if this is a bug or just not supported.
Starting a print via ESP3D (2nd serial port), for a file located on the motherboard USB/SD storage does not work.

### Description
Listing files of the motherboard storage with M20 works fine. But when pressing the print button on the ESP3D interface the print doesn't start, I just get an "ok" back.

### Steps to reproduce
1. Setup: Motherboard with onboard USB/SD Card, connected to the TFT (USB/SD supported enabled in Marlin), ESP3D wifi interface connected to the TFT.
2. Insert a USB drive or SD card in the motherboard and list the files via the ESP3D interface. This works fine.
3. Press the print button of one of the files on the ESP3D interface

**Expected behavior**
The printer should start printing the selected file.

**Actual behavior**
The file is selected, but the print is not started.

### Hardware Variant
MKS Monster8 V1.0, BTT TFT35 V3.0, ESP3D on TFT 2nd serial port

### TFT Firmware Version & Main Board Firmware details
Latest TFT firmware (Feb 18th 2022).
Marlin release 2.0.9.3 (Feb 18 2022).

### Additional Information
I can follow the process in the source code, interfaceCmd.c
M20 (list files) is correctly forwarded to the motherboard, but infoFile.source is not updated, it stays at the startup value: TFT_SD. It should probably be set to BOARD_SD.
M23 (select file) is correctly forwarded to the motherboard and works correctly. (correct Marlin response).
M24 (start/resume print) is not forwarded to the motherboard because "infoFile.source == TFT_SD", but the file that should be printed is not located on the TFT SD port.

digant73, 2022-02-21:
yes there is also a bug (wrong title in Printing menu) fixed in #2406 (although you currently never reach the Printing menu).
infoFile.source should be set to BOARD_SD_REMOTE.
I will check and fix your issue.

rondlh, 2022-02-21:
@digant73 I tried your current fork, but it doesn't solve my issue.
I can see that once M24 is called, infoFile.source is still TFT_SD, which means that the motherboard never gets the M24, but perhaps this is normal, I'm not sure...

digant73, 2022-02-21:
I'm fixing something in M23.
With M23 you should already receive something like:

`File opened: ANGOLO~1.GCO Size: 118995
`
and printRemoteStart function should be invoked in parseAck.c

rondlh, 2022-02-21:
Yes, I'm sure I'm sending "M23 <filename>".
I tried it via the ESP3D interface, first get the directory listing of the motherboard USB port, then press one of the available print buttons.
I did the same via the ESP3D console (M20, M23 <filename>, M24, but no luck. I put some debug statements in interfaceCmd.c, at the case 24:
There I can see that infoFile.source is still TFT_SD. 
In the case 23: nothing will happen because the command is just "M23 <filename>", there is no "M23 SD:" or "M23 U:"

rondlh, 2022-02-21:
Thanks again!
I have "onboard_sd:2", not sure about the menu, I didn't check the screen, only the messages on the ESP3D.
I will do some more testing later, the printer is busy now for 10 hours to crack out 3 Bing Dwen Dwen...

digant73, 2022-02-21:
just updated my PR with minor cleanup and code reduction. With M115 verify that you get Cap:SDCARD:1. If so, the following check in parseAck.c should be matched:

```
// parse and store M23, select SD file
else if (infoMachineSettings.onboardSD == ENABLED && ack_seen("File opened: "))
```

and the Printing menu opened.

I tried from "SD Files" and it's OK.
I tried the following commands from ESP3D "Commands" and it's also OK:

M20
M23 ARM~1.GCO     (only of the files listed by M20)
M24

With M23, the Printing menu is opened and moved to pause
With M24, the print is resumed/started

Try my last changes in my PR (but they should not solve your issue)

rondlh, 2022-02-22:
I tried your fork just now, and it seems to work fine, great job again. The capabilities are reported correctly, which I would expect because I can list the files. The TFT menu changes correctly. I use ESP3D 2.1 Gui. I think onboard_sd in config.ini was disabled previously.

digant73, 2022-02-22:
I think the issue (the hang after "Storage OK") is due to onboard USB. Maybe something is different compared to onboard SD. I should have the messages sent by Marlin just to understand what is done in the TFT side.

rondlh, 2022-02-22:
To test I disabled the USB (USB_FLASH_DRIVE_SUPPORT and USE_OTG_USB_HOST) in Marlin, then the motherboard SD Card is active. I can see the files via ESP3D, but the same issue happens when using the TFT print menu. "Storage OK" and the TFT hangs. The motherboard is still running. The Storage OK message seems to come from cardreader.cpp (Marlin).
Please let me know if you want me to do some tests...

digant73, 2022-02-22:
are you able to intercept on TFT (maybe with debug option) the messages exchanged with Marlin before the hang?

digant73, 2022-02-22:
The problem with no print after confirmation is due multi volume currently not supported. If you try with onboard SD (not USB) it should work

rondlh, 2022-02-23:
In Marlin I have MULTI_VOLUME disabled, so I assume that the motherboard SD card is not active. For Marlin SD or USB should be completely transparent I believe. Either way, I took your fork, and disabled USB in Marlin (USB_FLASH_DRIVE_SUPPORT, USE_OTG_USB_HOST). Now via the TFT I can see the files on the motherboard MicroSD port, but the problem stays the same, after selecting and confirming the print the TFT immediately returns to the directory listing without doing anything. I'm puzzled...

digant73, 2022-02-25:
try PR #2419

rondlh, 2022-02-26:
Downloaded your latest fork (updated on Feb 25th), tried again, but same result, back to file listing after print confirm. Tried with SD card only... anything you want me to try?

digant73, 2022-02-26:
In "gcode.c", edit "request_M21()" function and change "M21\n" with "M21 U" (if your files are in a remote USB) or M21 S (if your files are in SD card)

kisslorand, 2022-02-26:
It won't work because my PR in Marlin's github (https://github.com/MarlinFirmware/Marlin/pull/23780) is not merged yet.

digant73, 2022-02-26:
ok, but if U and S options are not yet supported by Marlin, what is currently the reason of the failure? is Marlin trying to access the wrong media?

kisslorand, 2022-02-26:
In Marlin mode you can switch the used (mounted) media.

kisslorand, 2022-02-26:
I just tested with current master to print a file located on SD Card in the motherboard and listing and selecting the file from ESP3D. The print started without any issue.
The same went without any trouble if the file was selected from the USB of the motherboard.

It seems there's no bug to fix.

digant73, 2022-02-26:
maybe you have PLR feature disabled. If PLR is enabled, the issue could be present (source for PLR file not set)

rondlh, 2022-02-26:
Yes, I have PLR disabled, let me switch it on an try again...

rondlh, 2022-02-26:
@kisslorand You are correct, you can start a print via the ESP3D interface, with both @digant73's fork and the current master code. The thing that was misleading me is that M109 and M190 now are blocking, and I guess also the BTT TFT start_gcode is not being executed anymore.
**I found another issue...**
Printing a file on the motherboard storage (SD/USB) via the TFT Print menu does not work. The screen returns to the file listing after confirming the print. If I manually send M20, then M23 <filename.gco> then the print starts correctly. No M24 is needed.
I will do some more testing...

digant73, 2022-02-26:
Usually I'm not so lucky and my TFT manifests some bugs that others don't have. But the bugs are there.
@rondlh, TFT menu calls M21 that you didn't reported as manually called. Try to call M21 first and see if you will have the issue as from the TFT menu

rondlh, 2022-02-26:
@digant73 So try: M21, M20, M23 <filename>?

digant73, 2022-02-26:
yes, and finally M24.
That's the standard sequence

rondlh, 2022-02-26:
I have no chance to enter M24 in the console, the screen jumps to the printingmenu after M23, and the Title says "Emulated M600", (M600 is for filament change) I have M600 emulation switched off in the TFT, also PLR is switched off.

digant73, 2022-02-27:
try my current project https://github.com/digant73/BIGTREETECH-TouchScreenFirmware. Use bugfix branch

rondlh, 2022-02-28:
> try my current project https://github.com/digant73/BIGTREETECH-TouchScreenFirmware. Use bugfix branch. That bug (wrong title in Printing menu) was fixed.

Works like a charm! Great job again, I hope your updates get merged soon.

rondlh, 2022-02-28:
Waiting for the code merge

PR #2419 (2022-03-21): Fixed remote printing bugs + code reduction and cleanup
**BUGFIXES:**
* Fixed wrong title in Printing menu if printing from remote: The bug can cause a freeze. Modified function getFilename (in vfs.h / .c) in order to provide the correct filename also if printing from remote (remote onboard or remote host). Furthermore, functions in vfs.c ordered according to vfs.h.
* Fixed PLR handling for remote printing from TFT (with M20-M24): PLR file could not be opened (missing check of source etc.) causing an error at print start (e.g. USB read error). Now PLR file is properly checked and opened. Furth [... cut]
closed 2022-02-28 by rondlh