bigtreetech/BIGTREETECH-TouchScreenFirmware #2423: [BUG] Rounding of parameter values

Guilouz, 2022-02-23:
In machine settings the 3rd digit after the decimal point is always rounded up or down.

For some parameters (like diagonal rod, diagonal rod trim, etc) we need to have precise values to 4 digits after the decimal point.

In example here we need to adjust settings with precise values (in red background).

[image]
 
Currently if I enter the value 314.8381, the result saved by TFT is 314.84 and they are saved in the EEPROM as is.

guruathwal, 2022-03-21:
I will look into it.

Guilouz, 2022-04-02:
@guruathwal Do you have checked this ?

guruathwal, 2022-04-02:
Yes, I have identified the Issue. I Will add a patch in a few days.

digant73, 2022-04-22:
@Guilouz 
To fix your issue it should be enough to apply the following changes in MachineParameters.c:

```
  {"H%.4f\n",            "S%.4f\n",       "R%.4f\n",       "L%.4f\n",      NULL,           NULL,           NULL,           NULL,           NULL,           NULL},           // Delta Configuration (Height, Segment per sec, Radius, Diagonal Rod)
  {"X%.4f\n",            "Y%.4f\n",       "Z%.4f\n",       NULL,           NULL,           NULL,           NULL,           NULL,           NULL,           NULL},           // Delta Tower Angle (Tx, Ty, Tz)
  {"A%.4f\n",            "B%.4f\n",       "C%.4f\n",       NULL,           NULL,           NULL,           NULL,           NULL,           NULL,           NULL},           // Delta Diagonal Rod Trim (Dx, Dy, Dz)
  {"X%.4f\n",            "Y%.4f\n",       "Z%.4f\n",       NULL,           NULL,           NULL,           NULL,           NULL,           NULL,           NULL},           // Delta Endstop Adjustments (Ex, Ey, Ez)

```

It is used **.4** instead of **.2**.

Guilouz, 2022-04-22:
I will try. Can you insert this in your actual PR ?

digant73, 2022-04-22:
It is better you try it and let me know if it solves your issue. Then I will add it on my PR

Guilouz, 2022-04-22:
Ok I will try tonight after work.

Guilouz, 2022-04-22:
@digant73 with 0.4, value are correctly registered in EEPROM but TFT always displays only 3 digits after the decimal point (although I can enter 4 then validate).

kisslorand, 2022-06-23:
Dear bot, the lack of activity is coming from @bigtreetech. We can't do much more than wait for @bigtreetech to merge the relevant bugfix.

PR #2505 (2022-07-12): Increase value display length after decimal in list view
- Increase value display length after the decimal point in list view.
- increase float precision for Delta configurations in **MachineParameters.c** and Numpad.c

Fixes #2423
closed 2022-07-07 by stale[bot]