bigtreetech/BIGTREETECH-TouchScreenFirmware #2427: [BUG] No Printer Attached and worse after update

Sonicboom247, 2022-02-28:
### Description

After updating to latest MKS TFT 28 V4 firmware on my Artillery Sidewinder X1 I am getting no Printer attached, Marlin firmware has baudrate set at 250000, selecting baudrate lower than 115200, restarts the printer. Also selecting LED on the TFT restarts the printer but not the TFT.

### Steps to reproduce

1. Select Menu Settings Connection and select any baudrate lower than 115200
2. [Second Step]
3. [and so on...]

**Expected behavior**
Test lower baudrate to remove no printer attached error

**Actual behavior**
Printer restarts

### Hardware Variant
MKS TFT 28 V4
### TFT Firmware Version & Main Board Firmware details
MKSTFT28V4 with BTT SKR 2 board and Marlin 2.0.9.3

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

Sonicboom247, 2022-02-28:
did the reset.txt to get back to the touch screen but back to No Printer attached, recompiled Marlin again with only 
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1  //UART

rondlh, 2022-02-28:
You will need to match the Marlin baud rate exactly otherwise the printer will receive some garbage characters which might trip up the printer (although I consider this to be very unlikely). I recommend to check your cabling. The TFT does a simple handshake with the motherboard and if the response is correct then the communication is established. "No printer attached" is a very poor choice of words, as in most cases that is not actually the problem.

Sonicboom247, 2022-02-28:
Thank you, i did verify the baudrate is set at 250000 on both Marlin and the TFT config.ini, i think if anything my wiring from TFT to board will be more of a culprit now that you mention it. Since the printer board was upgraded, I will pull and verify wiring in the morning

rondlh, 2022-02-28:
Do you know the old and new motherboard models?

Sonicboom247, 2022-02-28:
Yes the old one was an MKS GEN L V1 and the new one is a BTT SKR2 V2

radek8, 2022-02-28:
> Yes the old one was an MKS GEN L V1 and the new one is a BTT SKR2 V2

BTT SKR2 V2 ?
What's this news?

Sonicboom247, 2022-02-28:
> > Yes the old one was an MKS GEN L V1 and the new one is a BTT SKR2 V2
> 
> BTT SKR2 V2 ? What's this news?

is it news or new? New yes, News not really as many users swap their boards out for ones with more features.

Sonicboom247, 2022-02-28:
now i am wondering if it is better to use EXP1 +5V GND PE12 PE13 instead of the TFT riser or if that is even an option

rondlh, 2022-02-28:
The TFT connection to the MKS Gen-L V1 (2x4 pin header) is different than the BTT SKR2 V2 (5 pin JST-XH). You will need to make new cable to get this to work.

Sonicboom247, 2022-02-28:
> The TFT connection to the MKS Gen-L V1 (2x4 pin header) is different than the BTT SKR2 V2 (5 pin JST-XH). You will need to make new cable to get this to work.

The weird thing is i had made a new JST connector and had this working just fine for about 3 months with no issues.  then I added a BLTouch...i have the feeling i loosened a wire because that's when everything went sideways, i actually had the BLTouch working and did a 5 hour print. then the extruder started skipping which was due to the ribbon cable going bad when reinserting after BLTouch wiring, i replaced the ribbon and now the BLtouch and extruder worked and TFT only says no printer so i am currently making a new cable.

Sonicboom247, 2022-02-28:
yup, looks like it was a loose cable/connector... that being said, will i get more feature control like the Neopixels if i connect to EXP1 vs the TFT header?

closed 2022-02-28 by Sonicboom247