bigtreetech/BIGTREETECH-TouchScreenFirmware #2432: [BUG] TFT35 switches mode automatically when print is started, then nothing happens

Alpacad, 2022-03-06:
### Description
In Marlin Mode I start a print from onboard media on the TFT35, it switches automatically to Touch Mode. After G34 it begins to heat the bed and it stops right before auto bed levelling parking the nozzle for no apparent reason. Tried to launch a print directly from Touch Mode and the result is the same. Trying to cancel the print causes the TFT to freeze.

### Steps to reproduce

1. Select a file from media
2. Start the print
3. Wait until auto bed levelling procedure

Expected behavior
Auto bed levelling, start printing.

Actual behavior
Nozzle is parked and heaters shutted down.

### Hardware Variant
TFT35 V3
### TFT Firmware Version & Main Board Firmware details
Latest firmware build BIGTREE_TFT35_V3.0.27.x.bin + Marlin 2.0.9.3

### Additional Information
Attached video and photos of the issue. Included the config.ini file and Marlin Configuration.h

https://user-images.githubusercontent.com/82800736/156929551-ad96bccb-d485-4d52-8429-02fc22660754.mp4

[image]
[image]

### ZIP files
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8192875/config.zip)
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8192888/Configuration.zip)

radek8, 2022-03-06:
Do you have a correctly connected filament sensor?

Alpacad, 2022-03-06:
Should be, light switches on when filament is inserted, but I didn't try to check it with M119

radek8, 2022-03-06:
I haven't used Marlin printing in over a year. I only use touch mode.

I have tested printing in Marlin mode and I can confirm that after starting printing in Marlin mode, the TFT will switch to touch mode.
I think it has to do with the addition of support for printing via Octoprint.
When switching to Marlin mode, there should be no automatic switch to touch mode when printing starts.

radek8, 2022-03-06:
I have a solution for you.
Set in the config.ini file or in the TFT touch menu
serial_always_on: 0

Alpacad, 2022-03-07:
I'll try this as soon as I come home from work.
Hope it'll work, in that case I'll flash a new config.ini with default at 0; I'll let you know, thank you.

digant73, 2022-03-07:
Radek8's hint is the proper solution. There are also other cases where a popup will be displayed in Touch Mode (print paused, M600, M117 messages etc...) if "serial always on" is set to true

Alpacad, 2022-03-07:
No, that didn't work. Right before reaching the bed first layer temperature, the nozzle goes to park posizion and all heaters are shutted down. After that these messages are displayed, then nothing happens.

[image]
[image]
[image]

And again, trying to stop the print causes the TFT to freeze. After a reset, these messages are displayed:

[image]
[image]

radek8, 2022-03-07:
Can you try this FW?
[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8201263/BIGTREE_TFT35_V3.0.27.x.zip)

Alpacad, 2022-03-08:
I'll try to flash it when I get home.
Could sharing the Cobfiguration_adv.h help find the issue?
Even though I didn't change anything from the previous working build apart from rx_buffer and support for M73.

digant73, 2022-03-08:
I didn't understand what's the issue now. If you set "serial_always_on" to "0", no switch to Touch Mode is possible. If you are referring to other errors during a print in Marlin Mode, the issue is non in TFT fw.

Alpacad, 2022-03-14:
Sorry, I wasn't testing Marlin Mode. I checked and now is working correctly. I was worried about the sudden stop, but I seem to have found the issue.
I need to do some more tests, then I'll confirm and close.

Alpacad, 2022-03-18:
So, about the strange behaviour, it was the PID setting for the bed in the Marlin Ender 5 Plus config example. I generally have my own PID values for different range of temperatures, but I apply them in the filament custom GCode section in PrusaSlicer (that way I can use specific PID values for different materials).
Since I didn't have issues in the previous releases of Marlin (I just need to get the bed to temperature, then the new PID values are applied and stored) and also no error messages regarding a failed heating appeared on the screen, I didn't even thought it could be that. Maybe the room temperature has become colder than before? I don't know.
After I manually added my default PID settings for PLA, everything worked correctly.
We can close this issue.

PR #2446 (2022-03-24): Fix untimely mode switching while Marlin Mode is active
- Fix mode switching while Marlin Mode is active.
  Prevent mode switching when serial is always on and print is started from other remote hosts. #2432.
- Misc clean-up

Fixes #2432.
closed 2022-03-24 by bigtreetech