bigtreetech/BIGTREETECH-TouchScreenFirmware #2436: SKR2 and TFT35 E3 screen mirrored after firmware update. Can someone help me fix it?

rick090980, 2022-03-08:
I recently changed my motherboard to a BTT SKR2 and the LCD to a TFT35 E3 (with encoded on bottom right like the Ender). Everything was working fine but I wanted to try out the Round Miracle theme so I followed the update instructions using a 4gb SD card formatted to FAT32, moved the TFT35 folder, TFT35 E3 .bin, and Config.ini to the root of the card. When I insert the card and turn on my printer the update begins with the very first screen (with the small text) oriented correctly but as soon as it starts the rest of the update for fonts, bmp etc the text is mirrored. After it completes and I do the touch screen calibration the entire screen is mirrored. 

All the touch positions are correct as in if I touch print it goes to the file selection screen, movement goes to movement screen.
I've already tried changing "Rotate UI" in the config.ini to 1 (enabled), changed rotation on LCD itself (it rotates but is still mirrored)
Since none of those worked I tried changing the rotation in the marlin configuration.h file to mirror_x, and a few others. That setting doesn't seem to change anything? 

Then found the tft_io.h file in Marlin firmware and changed the section with user rotation to look like:

// TFT_ROTATION is user configurable
#ifndef TFT_ROTATION
#define TFT_ROTATION TFT_MIRROR_X
#endif

Again no change...

I'm seriously about to lose my mind trying to fix this!! I can't even get it back to how it was before I messed with it. 
I greatly appreciate any advice.

rondlh, 2022-03-08:
Sounds like you are flashing the wrong firmware...

digant73, 2022-03-08:
download and install the firmware from the **master** branch as reported in README.md

rick090980, 2022-03-08:
Yeah, that's where I got the firmware I'm currently using from. There's only one choice for the E3 version of the TFT35 so it'd be pretty hard to choose the wrong one. I've currently tried the 26 and 27 .bins. Wouldn't want to go back any further because I'd be losing a lot of functionality like the Auto Z-align function, which I use/need.

rick090980, 2022-03-08:
I even bought a new 4gb sd card and formatted it thinking maybe the other card I was using was bad,  but no dice... same problem. It's odd that when the update first begins, the very first screen, the text is oriented correctly, but mirrors from then on. I've read about others having the same or similar issue but none of them posted a fix.

digant73, 2022-03-08:
try to put a blank file named `reset.txt` in the root of the SD card and reboot the TFT. The calibration procedure will start

radek8, 2022-03-08:
Be precise. No "TFT35 E3 .bin" file exists
You used to update the old version.
Use the master!

radek8, 2022-03-08:
[image]

rick090980, 2022-03-08:
OH that might be it! I definitely chose the GD version a bit further up from the one you circled. I'll give that a go soon as I get home.

rick090980, 2022-03-08:
> 

I understand that TFT E3.bin doesn't exist I shortened it for ease. Just wanted to assure that I was using the version meant for the E3 version LCD and not the regular TFT35.

rick090980, 2022-03-09:
Well, unfortunately that wasn't it. I downloaded the firmware using that link before and it was the exact same files. The "GD" files aren't included in the source code zip I downloaded. I attached a pic that shows I am using the correct firmware, and the mirrored text. I think I'm just going to return my LCD as defective cause I'm totally lost.
[image]

radek8, 2022-03-09:
The problem is not in the display and in you.
You use old FW.
Your version is from April 1, 2021.
They keep telling you all here, you would download the current MASTER versions.

[image]

radek8, 2022-03-09:
The current version is 15 days old ....

radek8, 2022-03-09:
see
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2411

rick090980, 2022-03-09:
Could someone PLEASE post a link to the latest TFT35  E3 firmware? And possibly the Round Miracle Theme? When I click on the master branch, then go to releases at the bottom the latest is from April 2021. I see now what you all mean, sorry took me so long, but I guess I just can't navigate Github as good as I thought because I can't find a way to download any later firmwares. Thank you all for your help!

digant73, 2022-03-09:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

rick090980, 2022-03-09:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
> 

Yeah I must be stupid, from that link I can browse through the code but how do I actually download it?

[image]

radek8, 2022-03-09:
[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8216420/TFT35.zip)

rick090980, 2022-03-09:
Sorry 😞... Thanks a lot for your continued help. Hopefully this is it and I be out of your hair.

rick090980, 2022-03-10:
[image]

All fixed!!! Seriously sorry for the frustration. Very much appreciate the help, you guys are awesome for taking the time to reply and help me through my problem. I guess this can be closed but maybe leave it somewhere in case someone else has the same problem? Thanks again!!

closed 2022-03-11 by rick090980