bigtreetech/BIGTREETECH-TouchScreenFirmware #2441: [BUG] ESP3D command console problem

NOVAXIM, 2022-03-13:
### Description

The problem is that the M710 report is dumped into the file window (I don't have such a file). The screenshot shows the problem:
[image]

**Actual behavior**

Of course, this is not just an ESP3D (firmware or WEBUI) problem. I say this, because when I touch MENU, I only have `T: ...` reports on the console and everything seems to be fine in file window.

**Expected behavior**
It would be nice to fix the M710 write so that it doesn't get pushed into the file window.

### Hardware Variant
SKR-MINI-MZ V1.0 + TFT35-MZ V3.0
ESP3D have UI: 2.1b77 / FW: 2.1.0.c26 (latest stable)

### TFT Firmware Version & Main Board Firmware details

### Additional Question
In addition, I have a question if it would be difficult to send to the console longname file before shortname file!?
This would make the file list with long filenames in the file window.

rondlh, 2022-03-14:
It's a known problem that some messages can interfere with the file listings. There is a file list start and end command, in between should be the files and sizes, but anything that is send during this time will end up as a item in the file listing. You can see the details in the console, it would be useful to show this here. The M710 responses are all at the end of the listing, are items being adding even after the file listing should be done? If so then the file list end command was not received correctly, but that would mean that you also should get temperature reports in there.
Could you describe in more detail how this happens? Did you send M710 several times just after requesting the file listing?

About your question, the file listing is managed by Marlin, not by the TFT, you can update the Marlin code to provide the long filename first if you like, but ESP3D 2.1 cannot handle filenames with space, so this will cause problems for you.

digant73, 2022-03-14:
for long filenames, ESP3D should simply use option L with M20 (M20 L) as done in one of the current pending TFT's PRs.

NOVAXIM, 2022-03-17:
> Could you describe in more detail how this happens? Did you send M710 several times just after requesting the file listing?

The screenshot shown the idle state (reports are renewed automatically every 3-5 seconds). At this point, on the TFT screen, I have the main menu. The situation changes when I select the menu on TFT, then only the temperature is reported on the console and everything is shown well on the file panel (without the M710 …).
I refresh the file panel each time with the button on the file panel (the WEBUI procedure is performed). The problem is, depending on the state of the TFT screen, I am getting M710… garbage or not.

> About your question, the file listing is managed by Marlin, not by the TFT, you can update the Marlin code to provide the long filename first if you like, but ESP3D 2.1 cannot handle filenames with space, so this will cause problems for you.

I figured it is on the Marlin side. Thanks for the attention to the space in the name (I didn't know that).

NOVAXIM, 2022-03-17:
Please noted that the lines started with `T:`, `X:`, `FR:`, `echo:` are properly filtered out by WEBUI file panel.

When on TFT is main menu, then I have more reports. Why?

digant73, 2022-03-18:
TFT main menu queries Marlin for some stats. One of those query is M710 Fan Ctrl (responsible for the M710 report). Your Marlin is reporting some stats ACKs after sending the "OK" ACK for the TFT query (e.g. M710). It's not an issue in the TFT fw.

oldman4U, 2022-03-24:
@Msq001 May you want to close this ticket.

closed 2022-03-25 by bigtreetech