bigtreetech/BIGTREETECH-TouchScreenFirmware #2445: [BUG] (short description)

iryj95, 2022-03-15:
### Description

I have a TFT35 E3 V3 with an skr 1.3 and tmc2130 drivers.
My problem is that the screen doesn't seem to keep the set eeprom setting in, even without restarting. So I would set for example the driver current for z to 800mA and start a print then it will lose steps I will stop the print, go in the settings and the screen will show current as 200mA. The same goes for every current, and they will not even stay the same between prints. Also it will change the feedrates and such to what ever it feels like.
Before I had a problem with the speed changing from 100% to something less, but that seems to have been fixed on the latest update. Although that wasn't as bad, since I could still use the printer.

Firmware has right values for driver currents, feedrates and accelerations, but screen changes them to what ever it feels like.

### Steps to reproduce

1. Set current for z axis as 800mA
2. When prompted tell to save to eeprom, also used the terminal to use m500

**Expected behavior**
Current stays at 800mA, even when starting a print.

**Actual behavior**
Changes to what ever the screen decides..

### Hardware Variant
TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
TFT 0.27.x Feb 22 2022 and Marlin 2.0.9.3 on the SKR 1.3

### Additional Information
[image]
This is before starting a print
[image]
This is after, since the current is gone too low z starts to miss steps and I have to stop the print.

oldman4U, 2022-03-15:
Please update the title to something useful.

iryj95, 2022-03-15:
> Please update the title to something useful.

Oops, sorry. Kind of a first time trying to ask help, usually I can find someone with a similar problem that will help, but this seems so strage.

guruathwal, 2022-03-16:
These settings are not stored on TFT. The issue seems to be in your Marlin firmware. The TFT only displays the values it receives from the Marlin firmware.

gordo3di, 2022-03-16:
If you have the option uncommented, the driver current will lower in 50 mA increments if your drivers are overheating. Its a safety feature in marlin.

iryj95, 2022-03-16:
The driver currents are set as,

  `#if` HAS_TRINAMIC_CONFIG

  `#define` HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current

  `#define` INTERPOLATE      true

  `#if` AXIS_IS_TMC(X)
    #define X_CURRENT       850        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  X_CURRENT / 2  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16        // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1        // -1..0: Not chained. 1: MCU MOSI connected. 2: Next in chain, ...
    //#define X_INTERPOLATE  true      // Enable to override 'INTERPOLATE' for the X axis
    //#define X_HOLD_MULTIPLIER 0.5    // Enable to override 'HOLD_MULTIPLIER' for the X axis
  `#endif`

  `#if` AXIS_IS_TMC(Y)
    #define Y_CURRENT       850
    #define Y_CURRENT_HOME  Y_CURRENT / 2
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
    //#define Y_INTERPOLATE  true
    //#define Y_HOLD_MULTIPLIER 0.5
  #endif

  `#if` AXIS_IS_TMC(Z)
    #define Z_CURRENT       800
    #define Z_CURRENT_HOME  Z_CURRENT
    #define Z_MICROSTEPS     16
    #define Z_RSENSE          0.11
    #define Z_CHAIN_POS      -1
    //#define Z_INTERPOLATE  true
    //#define Z_HOLD_MULTIPLIER 0.5
  #endif

  `#if` AXIS_IS_TMC(E0)
    #define E0_CURRENT      800
    #define E0_MICROSTEPS    16
    #define E0_RSENSE         0.11
    #define E0_CHAIN_POS     -1
    //#define E0_INTERPOLATE true
    //#define E0_HOLD_MULTIPLIER 0.5
  #endif

I will look into the current step down, but that doesn't really explain why the acceleration values change from example x axis from 2000 set in marlin firmware to 9000 shown on the screen settings after stopping the print (and I am pretty sure that it is using them while printing also)

iryj95, 2022-03-17:
Ok, so I noticed it might be pulling the accelerations from prusaslicer, does this line "file_comment_parsing:1" on config have something to do with it?
Still weird that it uses the slicer settings and won't respect the firmware maximums :o, as for the currents, it probably was the overheating, as I was doing a g29 the terminal showed the overheat warning on z, so I lowered the current a little bit, and that seemed to help.

closed 2022-03-19 by iryj95