bigtreetech/BIGTREETECH-TouchScreenFirmware #2452: on start-up Z axis raises and BL-Touch  probe deploys

discip, 2022-03-21:
### Description
After the latest update (see below) the on power-up the **Z axis raises** and the **BL-Touch probe deploys**. 

### Steps to reproduce

1. Update to latest build (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/ac3c2afd136f1e4e4e68ad46a808acf2e0f709a0).
2. Switch off your printer.
3. Switch your printer back on.
4. Observe the printer to raise the **Z axis** about **`12mm`** and **deploying the BL-Touch probe**.

**Expected behavior**

After turning on the the printer nothing should move without the user interaction.

**Actual behavior**

See **`Steps to reproduce`**.

### Hardware Variant

MKS TFT28

### TFT Firmware Version & Main Board Firmware details

(https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/ac3c2afd136f1e4e4e68ad46a808acf2e0f709a0)

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8318915/Configuration.zip)

kisslorand, 2022-03-21:
Update to latest Marlin bugfix and it won't do that.
Upon startup the TFT makes a new check about a new Marlin's capability, the BLTouch High Speed mode.

discip, 2022-03-21:
## Thanks!
Yep, that worked! 👍
Should have checked before posting. 😓

kind regards

closed 2022-03-21 by discip

kisslorand, 2022-03-24:
#2459

PR #2459 (2022-03-26): Tiny fixes, tiny program size reduction
### Requirements

BTT or MKS TFT

### Description

- BLTouch HS mode button updates from Marlin's feedback, not from desired value
- tiny program size reduction
- stow BLTouch probe after initial BLTouch HS mode check (guard for older Marlin than 12.III.2022)
- some tiny HU language grammar fix

### Related Issues

Fixes #2458
Fixes #2454
Fixes #2452