bigtreetech/BIGTREETECH-TouchScreenFirmware #2454: [BUG] When the printer is powered on the BLTouch is deployed.

magnificu, 2022-03-23:
### Description
When the printer is powered on the BLTouch is deployed.
This can damage the probe if the used move any axis.
All this happens after commit 2b97218b76eb965647051f7cdbe43354334d63f0.

This behavior should not be hard coded.

**Expected behavior**
When the printer is powered on the probe should not be deployed.

rondlh, 2022-03-23:
Please provide all information requested by the bug report template.
The TFT doesn't do anything with the BLTouch during startup. What firmware are you using, if it's Marlin, then check your STARTUP_COMMANDS

kisslorand, 2022-03-23:
> The TFT doesn't do anything with the BLTouch during startup.

Actually it does. It checks if HS mode of BLTouch state. (M401 H)

magnificu, 2022-03-23:
Using the same Marlin version.

BIGTREE_TFT50_V3.0.27.x.bin -> from 10.03.2022 does not deploys the probe.
BIGTREE_TFT50_V3.0.27.x.bin -> from 21.03.2022 deploys the probe.

The problem is that M401 H deploys the probe when the HS mode is disabled in Marlin !!!
(//#define BLTOUCH_HS_MODE true)

Not all the users have/use HS mode.
Again, such a behavior should not be hard coded in the source code.

kisslorand, 2022-03-23:
@magnificu 
Your Marlin is not up to date, latest Marlin doesn't deploy the BLTouch with "M401 H" command.

magnificu, 2022-03-23:
Ok.
Thanks for the help.
When the TFT firmware requires a minimum Marlin version I guess a warning should be raised or at least a message on the printer display, but maybe as ask too much...

closed 2022-03-23 by magnificu

kisslorand, 2022-03-23:
It was you who mentioned commit https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/2b97218b76eb965647051f7cdbe43354334d63f0 from PR #2393. I guess you missed what is written in that pr so I copy & paste it here for you:

Requirements:
BTT or MKS TFT
a BL Touch bed levelling sensor or alike
**Marlin from 12.III.2022 or newer**
"#define BLTOUCH_HS_MODE true" in Marlin's "Configuration_adv.h"

sarvenn, 2022-03-24:
@kisslorand hi,
I could not find the option for HS mode on/off you implemented with #2393.
My marlin is up to date and HS mode is true. I expected to see this feature under BL touch menu under ABL where I can stow, deploy, test or reset my BL touch. But it wasn't there. Am I doing something wrong or missing something. I read the explanation but I must be missing something. Would you help?

kisslorand, 2022-03-24:
Send "M401 H" to Marlin and check if you got a response or not from Marlin.

sarvenn, 2022-03-24:
I will but isn't there a button to switch on/off in TFT mode? I understood like that from your PR.
Should I use the terminal to send Gcode for on and off command?

rondlh, 2022-03-24:
Sending a "M402" after the "M401 H" could be an easy workaround.
This seems to have been fixed in the latest Marlin 2.0.x bug fix. I personally never use the Marlin bug fixes, I only use the stable Marlin releases, the Marlin March 13th update does not support the M401 H option yet.

magnificu, 2022-03-24:
In general I find it very strange to implement features that are not supported by the stable releases.
I guess it is happening just because there is no sponsor/maintainer with a minimum strategy and vision.

kisslorand, 2022-03-24:
> In general I find it very strange to implement features that are not supported by the stable releases. I guess it is happening just because there is no sponsor/maintainer with a minimum strategy and vision.

What feature in the stable release of the TFT FW is not supported by Marlin's stable release? I am not aware of any so can you please enlighten us and point out what would be that feature that bothers you?

kisslorand, 2022-03-24:
#2459

magnificu, 2022-03-25:
The options about how to fix the "new feature" are clear, but some questions remain.
Why the average users should look for fixes after the TFT firmware update?
Why such a feature with potential to damage hardware is introduced without any notification or Marlin version check warning?

kisslorand, 2022-03-25:
Average users should stay away from nightly builds if they do not want to take any chance.
I do not know what you want other than ranting and fighting.

sarvenn, 2022-03-26:
> Yes, there's a button, but it is visible(active) only if HS mode is set to true in Marlin. All this only in Marlin from 12.III.2022. That's why I told you to send "M401 H" to check if your Marlin has the capability to report BLTouch HS mode state.

Button is missing.
[image]
[image]

kisslorand, 2022-03-26:
@sarvenn 
Are you sure you're on the latest BTT FW? I cannot reproduce the issue.

PR #2459 (2022-03-26): Tiny fixes, tiny program size reduction
### Requirements

BTT or MKS TFT

### Description

- BLTouch HS mode button updates from Marlin's feedback, not from desired value
- tiny program size reduction
- stow BLTouch probe after initial BLTouch HS mode check (guard for older Marlin than 12.III.2022)
- some tiny HU language grammar fix

### Related Issues

Fixes #2458
Fixes #2454
Fixes #2452